kalymos/PsNee #4: Improve instructions.

damopinn, 2017-06-28:
I used this but i commented out the NTSC fix because my ps1 doesn't need it. I think it would hang waiting for a signal it wont receive if I didn't. It doesn't mention you should edit the inject_playstation() for your region.

kalymos, 2017-06-29:
Hi if you use PsNee v6, test this version [https://github.com/kalymos/PsNee/blob/6ec387f30d0540b84bb129c454a168752bf1613f/PsNee.ino](https://github.com/kalymos/PsNee/blob/6ec387f30d0540b84bb129c454a168752bf1613f/PsNee.ino) It works with the same welding as the v6 (For me v6 does not work properly) There is also the psxdev.net that works well, seley it has not the same welding.

damopinn, 2017-06-29:
At the moment I'm not going to try I just rebuilt my ps1 with an attiny85 using v6. It's working OK.
Which region are you because v6 is Europe only but you can edit it for other regions.
and the NTSC fix needs to be removed if you don't need it I presume it will cause a hang, from reading the code it waits for an input from some bios pins which I didn't connect. What is your problem with v6?

kalymos, 2017-07-01:
I'm on the European region with SCPH-102, normally I need NTSC patch. I tested practically all the versions and it is the only one that I failed to run.
I tried several manipulation on the code, remove the patch NTSC, and something else. She always refused to inject code on the data line.

damopinn, 2017-07-01:
What device are you using because I used the Attiny85. the pin numbers are Arduino pin number like on a Digispark. using a pro mini is a lot easier. I can send you my edited code if you want.

kalymos, 2017-07-01:
I use a mini pro, and I tested an Attiny 45. I want your code, it will allow me to see where I follow myself.
I also have the new psxdev code to test.

damopinn, 2017-07-01:
I made a fork and uploaded the file. I can help to test when I'm not too busy. I suppose you leave a header to make it easier to test

kalymos, 2017-07-03:
It's good its function I had to go like a dirty :)