kalymos/PsNee #5: tried to port to attiny

damopinn, 2017-07-05:
I added some code to the psxdev program. 

  #define ATTINY

    #ifdef ATTINY
      // board pins
      #define sqck 3          // connect to PSX HC-05 SQCK pin
      #define subq 4          // connect to PSX HC-05 SUBQ pin
      #define data 2          // connect to point 6 in old modchip diagrams
      #define gate_wfck 0     // connect to point 5 in old modchip diagrams
      // MCU input / output
      #define SUBQPORT PINB       // Atmel MCU port for the 2 SUBQ sampling inputs
      #define SQCKBIT 3           // ATmega PD6 "SQCK" Mechacon pin 26 (PU-7 and early PU-8 Mechacons: pin 41)
      #define SUBQBIT 4           // ATmega PD7 "SUBQ" Mechacon pin 24 (PU-7 and early PU-8 Mechacons: pin 39)
      #define GATEWFCKPORT PINB   // Atmel MCU port for the gate input (used for WFCK)
      #define DATAPORT PORTB      // Atmel MCU port for the gate input (used for WFCK)
      #define GATEWFCKBIT 0       // ATmega PB1
      #define DATABIT 2          // ATmega PB0
    #endif

on the attiny there is only portb and the bit number is the same as the pin number so I changed pind to pinb and the pins and mcu input output to match but it doesnt work.

kalymos, 2017-07-07:
I would not lie. It seems to me that there is a registry call which does not conform with the Attiny. Most people even to answer exactly, are on psxdev. I do not really develop the code, I test it within the limit of my abilities ;)

damopinn, 2017-07-08:
Thanks, I found the thread on psxdev. I've been chatting there a few days now. Good job on putting this on github

closed 2023-04-28 by kalymos