kalymos/PsNee #9: Arduino Pro Micro support?

gamelaster, 2017-08-28:
Hey, I have SCPH-5502, trying to use Pro Micro for modchip, but sooo, what to do, since: `has a different pin assignment and needs some easy porting`. Any more information about it?

Also, how find out what PU I have?

Thanks

kalymos, 2017-08-28:
Hi if your motherboard looks like this [-->](https://upload.wikimedia.org/wikipedia/commons/thumb/b/b8/Sony_Playstation_1_SCPH-5502_motherboard_top.jpg/773px-Sony_Playstation_1_SCPH-5502_motherboard_top.jpg) It is a PU-18, you will find an adequate diagram in PsNee-V7-Diagram.

ramapcsx2, 2017-08-29:
Pro Micros can do the job but the ATmegaU chip has differnet ports and the Arduino pinout changed.
Since this code uses direct port manipulation and the pinout changed, you will have to find free pins and tell the code to use them.
For example, 'data' uses PB0, assigned to pin#8 on Arduino.
On Pro Micro, PB0 isn't available (used for an LED), so the data pin has to move and all port manipulation adjusted accordingly.
If you don't feel like doing all that, get a Nano. That should be compatible with the Arduino Uno #define ;)

ramapcsx2, 2017-08-30:
Those pins look good, yeah.
It looks like you can re-use all port defines from 'ARDUINO_BOARD'. Nice choice of pins :)
All you'll have to change is the pin numbers (sqck, subq, data, gate_wfck) and the bits (SQCKBIT, SUBQBIT, GATEWFCKBIT, DATABIT).
This should already work then.
Solder the wires, connect Vcc and ground as per the installation diagrams (now with your changed pins) and give it a test!

gamelaster, 2017-08-30:
@ramapcsx2 well, I probably already have mod chipped so I don't know if I want to do more work on it, but for contribution, I will probably try that! I will post here the results (Serial dumps if it will not work) later 😊 
Thank you for now

LameNobody, 2017-10-03:
Well, tested it out with these pins, 
`SQCKBIT => pin 2 (PD1)`
`SUBQBIT => pin 3 (PD0)`
`GATEWFCKBIT => pin 15 (PB1)`
`DATABIT => pin 14 (PB3)`
It booted the backup.. so I'm going to assume it worked fine.

kpek, 2020-03-22:
can anyone who got this to work with Pro Micro please share what exactly has to be changed in the code ? when i read all the info in this thread its confusing, 
Do we change the pin defines only ? or only bits defines ? or both  ?

Thanks

wyatt8740, 2020-05-25:
@kpek check pull request #22 . I have this installed inside my SCPH-1001 on a pro micro.

Clone [my repository](https://github.com/wyatt8740/PsNee) and set `#define ARDUINO_32UX_BOARD`. My repository has the readme updated with the correct pinouts for it.

wyatt8740, 2020-05-28:
@ramapcsx2 can this issue be closed now since PR #22 was merged?

gamelaster, 2020-05-28:
@wyatt8740 I will close it. Thank you for testing :)

BTW. I was trying it too, sadly, after all soldering, code changes, I dummy forget that I have 5V version of ProMicro 🤦

closed 2020-05-28 by gamelaster

wyatt8740, 2020-05-29:
> BTW. I was trying it too, sadly, after all soldering, code changes, I dummy forgot that I have 5V version of ProMicro
I have the 5V pro micro as well. It still worked for me, just feed it 3.5V on the Vcc pin or whatever it's called. The pin that doesn't go through a regulator. It might actually be called `5V`, but connect 3.5V to it anyway.

gamelaster, 2020-05-29:
@wyatt8740 Well, I thought the reason why it doesn't work is because 5V, well I had issue somewhere else! Good to know!

wyatt8740, 2020-06-04:
@gamelaster Glad it helped. If your board has LED's and they just turn on and stay dimmer than usual it's usually a sign they're being powered wrong through the regulator. But even though AVR's aren't really designed to run at 16MHz on ~3.3V in practice it seems to work fine with the 3.5V from a PSX at least.

Took me a while to figure that out, too, but I think rama3 or someone else on the psxdev forum thread brought it up.