kalymos/PsNee #10: PSone Pins

Madad1996, 2017-10-04:
Is this compatible to the PSone, if yes which pins i should use?

ramapcsx2, 2017-10-05:
PAL or NTSC-U/J machine?
PAL requires the boot patch, which means more cables.
In any case, the install diagrams include this model. The diagram is for PAL.

Madad1996, 2017-10-06:
Cant find the Pins on the Board...
https://imgur.com/gallery/9Iwq6

Madad1996, 2017-10-06:
SCPh-102

ramapcsx2, 2017-10-07:
Ah, this is the more rare PM-41(2) board. 
You'll have to use a multimeter and find something to solder to. Google image search has some old modchip installation pictures for these boards. They can help you find your pins.
The 2 BIOS pins will be the same, so will Vcc and Gnd.
Data is usually marked '6' in old diagrams. It will be one side of an smd capacitor. 
WFCK is marked '5' and is a test point pad.
SUBQ is pin24, SQCK is pin26 on the HC-05 (IC304).

closed 2023-04-28 by kalymos