kalymos/PsNee #11: non-working games

zoinknoise, 2018-01-27:
i apologize if this isn’t the right place to report this. have found a few non-working games.

-Henry Explorers (SLPM-86021) boots to “Insert PlayStation cd-rom”
-Tokimeki Memorial: Forever with You (SLPS-00065) boots to CD player

both are Konami games from 1995, strange coincidence.

i also found a game that trips the anti-mod protection, though I was booting a copy not an original: Ore no Ryouri (SCPS-10099) this game was burned from a Redump image.

my setup is SCPH-1001, PU-8 board, ACM drive. using latest version of code, modified a/a/a boot string on Arduino pro mini clone @16mhz.

zoinknoise, 2018-01-27:
update: uninstalled PsNee and replaced with Mayumi v1.1 (PICF508). can confirm these three games boot successfully. but now Spyro 3 doesn't work, oh well. :P 

just reporting, cuz i like this project. :)

ramapcsx2, 2018-01-27:
Thanks :)
I can imagine what the failing games do. The non-booting titles are probably using a rare disc layout, maybe lots of audio in the TOC.
The one that trips the antimod, not sure. That should be almost impossible.

ramapcsx2, 2018-01-27:
Ore no Ryouri is working just fine on either a PAL (expected) or Japan PSX (should fail, if the code was bad).
The check is right after the SCE message on bootup, right? (I can hear the laser move, to search for modchip symbols. Then it continues fine, meaning it passed the test.)

zoinknoise, 2018-01-27:
hmm... i didn’t try more than once, maybe i should have. it certainly could have been  the extremely finicky ACM drive causing the error. mine is in good condition but it still despises CD-R’s. 

my machine is NTSC U/C region btw.

zoinknoise, 2018-01-27:
yes the error happened immediately for me after boot. the screen actually came up in a Japanese which I can’t remember if that’s typical for a US machine.

zoinknoise, 2018-01-29:
maybe there can be an option to force game mode? so if the disc is being mistakenly booted as audio, there can be a manual override. just an idea~

zoinknoise, 2018-01-30:
reinstalled PsNee. can confirm Ore no Ryouri works fine. must have been a fluke.

trying more Konami games and found another that won't boot:

-Sexy Parodius (SLPM-86009) boots to "insert PlayStation CD-ROM"

not an original, but burned multiple times from Redump to confirm.

ramapcsx2, 2018-01-30:
The code should just work, no options necessary. If there are games that don't start, I consider it a bug that needs fixing :)
I actually have Parodius. I'll see what it does here.

ramapcsx2, 2018-02-01:
No problem on Parodius either. Check your wiring, try to attach a serial monitor and read the debug messages.

zoinknoise, 2018-02-01:
are we talking about the same Parodius? there are three for PS1 i believe. Sexy Parodius is the one that won't boot; the other two will.

if all three should work, then it must have to do with my PS1's issue with CD-R discs since i don't own original copies of those. but the two games at the top, Tokimeki and Henry Explorers, do not boot from spotless original copies. they DO boot with a Mayumi on the exact same machine/drive, so i must assume the discs are not defective.

besides these few games, i have had total success booting all backups and originals with the current version of PsNee! i will get things wired up and post some serial logs soon.

ramapcsx2, 2018-02-01:
Okay, I found Henry Explorers and this one really doesn't work.
If you look at the logs, it's easy to tell why:
61 00 A0 99 35 28 00 01 20 00 FC FF 
The game has one bit too many set, compared to all other titles.
Adding the flag fixes the problem.

zoinknoise, 2018-02-01:
sweet! will the next version of the code have this fix?

i will test and log Tokimeki and Sexy Parodius later today if you don’t get to them first.

ramapcsx2, 2018-02-02:
Yeah, this is going to need a new revision. I think I also found a nicer way to go about the injection timing.
Normally, this (and old codes) use a fire and forget method to unlock the HC-05.
The new method looks at the effects of a single injection and can break/exit as soon as it thinks the HC-05 has been unlocked.
This is all to try and reduce interference to normal operation, since the modchip action is pretty bad for the error correction and thus, the laser.

zoinknoise, 2018-02-02:
excellent to hear! here are the logs i promised:

zoinknoise, 2018-02-02:
Tokimeki Memorial: Forever with You [SLPS-00065]

MCU frequency: 16000000 Hz
Waiting for SQCK..
highs: 4832 lows: 0
pu22mode: 0
21 00 A2 06 39 26 00 70 38 11 69 1F 
61 00 01 06 39 31 00 00 02 00 FC FF 
21 00 02 06 39 33 00 66 24 14 9F 7F 
21 00 02 06 39 34 00 66 24 14 FC FF 
61 00 A0 06 39 35 00 01 20 00 9F 7F 
61 00 A0 06 39 36 00 01 20 00 FC FF 
61 00 A0 06 39 37 00 01 20 00 9F 7F 
01 00 A1 06 39 38 00 02 00 00 FC FF 
01 00 A1 06 39 39 00 02 00 00 9F 7F 
01 00 A1 06 39 40 00 02 00 00 FC FF 
21 00 A2 06 39 41 00 70 38 11 9F 7F 
21 00 A2 06 39 42 00 70 38 11 FC FF 
21 00 A2 06 39 43 00 70 38 11 9F 7F 
61 00 01 06 39 44 00 00 02 00 FC FF 
61 00 01 06 39 45 00 00 02 00 9F 7F 
61 00 01 06 39 46 00 00 02 00 FC FF 
21 00 02 06 39 47 00 66 24 14 9F 7F 
21 00 02 06 39 48 00 66 24 14 FC FF 
21 00 02 06 39 49 00 66 24 14 9F 7F 
61 00 A0 06 39 50 00 01 20 00 FC FF 
61 00 A0 06 39 51 00 01 20 00 9F 7F 
61 00 A0 06 39 52 00 01 20 00 FC FF 
01 00 A1 06 39 53 00 02 00 00 9F 7F 
01 00 A1 06 39 54 00 02 00 00 FC FF 
01 00 A1 06 39 55 00 02 00 00 9F 7F 
21 00 A2 06 39 56 00 70 38 11 FC FF 
21 00 A2 06 39 57 00 70 38 11 9F 7F 
21 00 A2 06 39 58 00 70 38 11 FC FF 
61 00 01 06 39 59 00 00 02 00 9F 7F 
61 00 01 06 39 60 00 00 02 00 FC FF 
61 00 01 06 39 61 00 00 02 00 9F 7F 
21 00 02 06 39 62 00 66 24 14 FC FF 
21 00 02 06 39 63 00 66 24 14 9F 7F 
21 00 02 06 39 64 00 66 24 14 FC FF 
61 00 A0 06 39 65 00 01 20 00 9F 7F 
61 00 A0 06 39 66 00 01 20 00 FC FF 
61 00 A0 06 39 67 00 01 20 00 9F 7F 
01 00 A1 06 39 68 00 02 00 00 FC FF 
01 00 A1 06 39 69 00 02 00 00 9F 7F 
01 00 A1 06 39 70 00 02 00 00 FC FF 
21 00 A2 06 39 71 00 70 38 11 9F 7F 
21 00 A2 06 39 72 00 70 38 11 FC FF 
21 00 A2 06 39 73 00 70 38 11 9F 7F 
61 00 01 06 39 74 00 00 02 00 FC FF 
61 00 01 06 40 00 00 00 02 00 9F 7F 
61 00 01 06 40 01 00 00 02 00 FC FF 
21 00 02 06 40 02 00 66 24 14 9F 7F 
21 00 02 06 40 03 00 66 24 14 FC FF 
21 00 02 06 40 04 00 66 24 14 9F 7F 
61 00 A0 06 40 05 00 01 20 00 FC FF 
61 00 A0 06 40 06 00 01 20 00 9F 7F 
61 00 A0 06 40 07 00 01 20 00 FC FF 
01 00 A1 06 40 08 00 02 00 00 9F 7F 
01 00 A1 06 40 09 00 02 00 00 FC FF 
01 00 A1 06 40 10 00 02 00 00 9F 7F 
21 00 A2 06 40 11 00 70 38 11 FC FF 
21 00 A2 06 40 12 00 70 38 11 9F 7F 
21 00 A2 06 40 13 00 70 38 11 FC FF 
61 00 01 06 40 14 00 00 02 00 9F 7F 
61 00 01 06 40 15 00 00 02 00 FC FF 
61 00 01 06 40 16 00 00 02 00 9F 7F 
21 00 02 06 40 17 00 66 24 14 FC FF 
21 00 02 06 40 18 00 66 24 14 9F 7F 
21 00 02 06 40 19 00 66 24 14 FC FF 
[... 618 lines cut ...]
61 01 00 00 00 05 00 00 01 69 9F 7F 
61 01 00 00 00 04 00 00 01 70 FC FF 
61 01 00 00 00 03 00 00 01 71 9F 7F 
61 01 00 00 00 02 00 00 01 72 FC FF 
61 01 00 00 00 01 00 00 01 73 9F 7F 
61 01 00 00 00 09 00 00 01 65 FC FF 
61 01 00 00 00 08 00 00 01 66 9F 7F 
61 01 00 00 00 07 00 00 01 67 FC FF 
61 01 00 00 00 06 00 00 01 68 9F 7F 
61 01 00 00 00 05 00 00 01 69 FC FF 
61 01 00 00 00 04 00 00 01 70 9F 7F 
61 01 00 00 00 03 00 00 01 71 FC FF 
61 01 00 00 00 02 00 00 01 72 9F 7F 
61 01 00 00 00 01 00 00 01 73 FC FF 
61 01 00 00 00 09 00 00 01 65 9F 7F 
61 01 00 00 00 08 00 00 01 66 FC FF 
61 01 00 00 00 07 00 00 01 67 9F 7F 
61 01 00 00 00 06 00 00 01 68 FC FF 
61 01 00 00 00 05 00 00 01 69 9F 7F 
61 01 00 00 00 04 00 00 01 70 FC FF 
61 01 00 00 00 03 00 00 01 71 9F 7F 
61 01 00 00 00 02 00 00 01 72 FC FF 
61 01 00 00 00 01 00 00 01 73 9F 7F 
61 01 00 00 00 09 00 00 01 65 FC FF 
61 01 00 00 00 08 00 00 01 66 9F 7F

zoinknoise, 2018-02-02:
Sexy Parodius [SLPM-86009] *burned from Redump*

MCU frequency: 16000000 Hz
Waiting for SQCK..
highs: 4959 lows: 0
pu22mode: 0
61 00 A2 99 50 04 00 47 32 64 82 7A 
61 00 A2 99 50 05 00 47 32 64 FC FF 
61 00 A2 99 50 06 00 47 32 64 9F 7F 
61 00 01 99 50 07 00 00 02 00 FC FF 
61 00 01 99 50 08 00 00 02 00 9F 7F 
61 00 01 99 50 09 00 00 02 00 FC FF 
61 00 A0 99 50 10 00 01 20 00 9F 7F 
61 00 A0 99 50 11 00 01 20 00 FC FF 
61 00 A0 99 50 12 00 01 20 00 9F 7F 
61 00 A1 99 50 13 00 01 00 00 FC FF 
61 00 A1 99 50 14 00 01 00 00 9F 7F 
61 00 A1 99 50 15 00 01 00 00 FC FF 
61 00 A2 99 50 16 00 47 32 64 9F 7F 
61 00 A2 99 50 17 00 47 32 64 FC FF 
61 00 A2 99 50 18 00 47 32 64 9F 7F 
61 00 01 99 50 20 00 00 02 00 FC FF 
61 00 01 99 50 21 00 00 02 00 9F 7F 
61 00 A0 99 50 22 00 01 20 00 FC FF 
61 00 A0 99 50 23 00 01 20 00 9F 7F 
61 00 A0 99 50 24 00 01 20 00 FC FF 
61 00 A1 99 50 25 00 01 00 00 9F 7F 
61 00 A1 99 50 26 00 01 00 00 FC FF 
61 00 A1 99 50 27 00 01 00 00 9F 7F 
61 00 A2 99 50 28 00 47 32 64 FC FF 
61 00 A2 99 50 29 00 47 32 64 9F 7F 
61 00 A2 99 50 30 00 47 32 64 FC FF 
61 00 01 99 50 31 00 00 02 00 9F 7F 
61 00 01 99 50 33 00 00 02 00 FC FF 
61 00 A0 99 50 34 00 01 20 00 9F 7F 
61 00 A0 99 50 35 00 01 20 00 FC FF 
61 00 A0 99 50 36 00 01 20 00 9F 7F 
61 00 A1 99 50 37 00 01 00 00 FC FF 
61 00 A1 99 50 39 00 01 00 00 9F 7F 
61 00 A2 99 50 40 00 47 32 64 FC FF 
61 00 A2 99 50 41 00 47 32 64 9F 7F 
61 00 A2 99 50 42 00 47 32 64 FC FF 
61 00 01 99 50 43 00 00 02 00 9F 7F 
61 00 01 99 50 44 00 00 02 00 FC FF 
61 00 01 99 50 45 00 00 02 00 9F 7F 
61 00 A0 99 50 46 00 01 20 00 FC FF 
61 00 A0 99 50 47 00 01 20 00 9F 7F 
61 00 A0 99 50 48 00 01 20 00 FC FF 
61 00 A1 99 50 49 00 01 00 00 9F 7F 
61 00 A1 99 50 50 00 01 00 00 FC FF 
61 00 A1 99 50 51 00 01 00 00 9F 7F 
61 00 A2 99 50 52 00 47 32 64 FC FF 
61 00 A2 99 50 53 00 47 32 64 9F 7F 
61 00 01 99 50 55 00 00 02 00 FC FF 
61 00 01 99 50 56 00 00 02 00 9F 7F 
61 00 01 99 50 57 00 00 02 00 FC FF 
61 00 A0 99 50 58 00 01 20 00 9F 7F 
61 00 A0 99 50 60 00 01 20 00 FC FF 
61 00 A1 99 50 61 00 01 00 00 9F 7F 
61 00 A1 99 50 62 00 01 00 00 FC FF 
61 00 A1 99 50 63 00 01 00 00 9F 7F 
61 00 A2 99 50 64 00 47 32 64 FC FF 
61 00 A2 99 50 65 00 47 32 64 9F 7F 
61 00 A2 99 50 66 00 47 32 64 FC FF 
61 00 01 99 50 67 00 00 02 00 9F 7F 
61 00 01 99 50 68 00 00 02 00 FC FF 
61 00 01 99 50 69 00 00 02 00 9F 7F 
61 00 A0 99 50 70 00 01 20 00 FC FF 
61 00 A0 99 50 71 00 01 20 00 9F 7F 
61 00 A0 99 50 72 00 01 20 00 FC FF 
[... 1023 lines cut ...]
61 01 00 00 00 08 00 00 01 66 FC FF 
61 01 00 00 00 07 00 00 01 67 9F 7F 
61 01 00 00 00 06 00 00 01 68 FC FF 
61 01 00 00 00 05 00 00 01 69 9F 7F 
61 01 00 00 00 04 00 00 01 70 FC FF 
61 01 00 00 00 03 00 00 01 71 9F 7F 
61 01 00 00 00 02 00 00 01 72 FC FF 
61 01 00 00 00 01 00 00 01 73 9F 7F 
61 01 00 00 00 00 00 00 01 74 FC FF 
61 01 00 00 00 08 00 00 01 66 9F 7F 
61 01 00 00 00 07 00 00 01 67 FC FF 
61 01 00 00 00 06 00 00 01 68 9F 7F 
61 01 00 00 00 05 00 00 01 69 FC FF 
61 01 00 00 00 04 00 00 01 70 9F 7F 
61 01 00 00 00 03 00 00 01 71 FC FF 
61 01 00 00 00 02 00 00 01 72 9F 7F 
61 01 00 00 00 01 00 00 01 73 FC FF 
61 01 00 00 00 00 00 00 01 74 9F 7F 
61 01 00 00 00 08 00 00 01 66 FC FF 
61 01 00 00 00 07 00 00 01 67 9F 7F 
61 01 00 00 00 06 00 00 01 68 FC FF 
61 01 00 00 00 05 00 00 01 69 9F 7F 
61 01 00 00 00 04 00 00 01 70 FC FF 
61 01 00 00 00 03 00 00 01 71 9F 7F 
etc.... boots to Insert PlayStation CD-ROM

ramapcsx2, 2018-02-03:
Tokimeki Memorial looks annoying.
It changes the control field from "data" to "audio" disk and can't quite decide whether "copying" should be allowed or not.
Anyway, I can catch it via the data flag.
Parodius is the same as Henry Explorers, just uses a flag that I didn't expect and never saw in tests.

zoinknoise, 2018-02-03:
awesome! it took a lot of effort to get you this data :)

ramapcsx2, 2018-02-03:
Yeah, I know.
We could use an Arduino Nano type device that has a USB port but doesn't power the device when USB is connected.
Alternatively, it could power just the USB bridge chip with the required 5V, and the rest of the board with 3.3V.

As it is, getting to the logs is pretty difficult when using a small internal Arduino.

ramapcsx2, 2018-02-03:
By the way, I'm working on the code. It looks like this'll take a while longer, so I think I'll put out an update to the current version with just the game fixes, then push my current work when it works :p

zoinknoise, 2018-02-03:
can’t wait!! :D

ramapcsx2, 2018-02-03:
Busy busy, but here's a couple fixes. It should work fine:
https://github.com/ramapcsx2/PsNee/commits/master

zoinknoise, 2018-02-03:
yes indeed! will test and report back.

zoinknoise, 2018-02-03:
just tested the new branch. all three games now work!! about time too, cuz now i can play Sexy Parodius ( ͡°з ͡°)

seriously though, this is great work. thank you for going to the trouble. i want this to be the very best project of its kind, so i’m glad i helped out a little.

finally a modchip for my PU-8, a board i really like, that boots ALL my weird games!

closed 2018-02-03 by zoinknoise

ramapcsx2, 2018-02-03:
Glad you like it :)