kalymos/PsNee #12: Any reason why we shouldn't use 5v Arduino?

mylogon341, 2018-01-27:
Hi, I'm _fairly_ new to electronics. I have some 5v Arduino Pro boards and found that these run on 3.3/3.5v no problem. Is there any reason why I shouldn't use 5v boards?
Thanks

zoinknoise, 2018-01-28:
that’s the kind of Arduino I’m using for PsNee with no problem, the 8mhz option is probably just for people with that board

ramapcsx2, 2018-01-28:
Every 5V Arduino I had / I tested ran fine at 3.3V / 16Mhz. But some of them will not rewrite correctly at that voltage. 
I'd say it is fine to flash a 5V / 16Mhz Arduino at 5V, and then use in the PSX at 3.3V.
Never had any issues and the system isn't critical.

mylogon341, 2018-01-28:
That's great to know. Thank you so much. I was just really confused on why there was such an emphasis on not using the 5v versions without any reason given. Good to know that I don't have to go out and buy more Arduinos.
Thanks

ramapcsx2, 2018-01-28:
The warning is a little different:
It says you must not use an Arduino *that is powered by 5V*, the reason being that it would send 5V level signals into the 3.3V PSX.

mylogon341, 2018-01-28:
Ah. So it does. I seemed to have misread/ misunderstood the point. That makes total sense.
Thanks @ramapcsx2 !

closed 2018-01-28 by mylogon341