kalymos/PsNee #15: Attiny85

gtchandra, 2019-01-01:
Hi, thanks for the great page, the project is awesome, I tried today to mod my father PS1 with an ATTINY85 and did not work.
I initially set the clockspeed at 8Mhz internal clock (burning the bootloader) then I programmed the chip using my programmer and an arduino board with ISP.
I also tried to get some debug details, connecting on pin3 @115200 and it seems working, injection is called sometimes, but I had no luck in making a burnt CD working (already tried on other modded boards). It seems injection is not effective. 
Any advice?
thanks

kalymos, 2019-01-01:
From memory, the stability of the ATTINY85 clock and problematic. The best solution is to use an arduino pro mini.

Or the https://github.com/kalymos/PsNee/blob/AttyNee/PsNee.ino branch, and built for ATTINY.

For more explanation the forum http://www.psxdev.net/forum/viewtopic.php?f=47&t=1262&start=480 is the best place.

happy New Year.

gtchandra, 2019-01-01:
Thank you! and happy New Year to you!!

gtchandra, 2019-01-01:
I'm trying to change code to http://www.psxdev.net/forum/viewtopic.php?f=47&t=1262&start=480
but I can't find all the necessary board endpoint. I was able to find LID, GATE and Reset using this diagram.
http://modchip.aeug.org/install/508-pu18-stealth.html
but still cant find 
// - PB3 = biosA18  
// - PB4 = biosD2   
on the PU-18 Board...

kalymos, 2019-01-01:
I understand why you can not find, the indication is in the code.

unsigned char biosA18 = 11; //Only used in SCPH-102 PAL mode

unsigned char biosD2 = 12; //Only used in SCPH-102 PAL mode

Thank you for the question, I just added the warning in the table.

ramapcsx2, 2019-01-08:
The chip works just fine, but your laser/drive/electronics have lots of trouble reading the disk.
Note: Once you see the black PS screen with a burned copy, you know that a modchip is working ;)

gtchandra, 2019-01-08:
that's what I thought, I have to clean and repair the optical drive.
Thanks, I'll let you know and eventually switch back to attiny85 (which worked as well up to that point) so we can put a TESTED on the SCPH5502.

kalymos, 2019-01-08:
Great thank you for keeping me informed.

gtchandra, 2019-03-02:
Hi there, just a quick one to update on the issue here.
The PSX I was working SCPH5502 on had serious issues related to the CD DSP. I tried on a SCPH 7502 (PU-22) with an ATTINY85 and it works perfectly. 
Thanks for your support.

closed 2019-03-02 by gtchandra