kalymos/PsNee #19: A problem with PM-41 NTSC-J boards 1-679-335-61

celcion, 2020-02-16:
I recently got a bunch of Japanese PS Ones (PM-41) sold in one lot as unchecked and while I was testing and fixing them, I also tried to mod them with PsNee. Most of them worked fine, but two boards with "1-679-335-61" printed on top of them just didn't want to boot CD-Rs no matter what I did. It starts spinning the disc, but not speeding it up for actual checking. And after some time it either shows the standard "Please Insert Playstation Format disc" screen, or switch to CD Audio player if the game on CD-R contains audio tracks. No injecting happens during this time either. As if I inserted the wrong disc.
Still, licensed discs worked fine and PsNee debug output showed that it does injecting. I also messed around a bit with injection making licensed discs not loading too as a test, so I determined that PsNee indeed does injection and PS One takes it. I also tried to change the CD drive to the one that worked fine in the other modded PS One and it didn't help too. These "1-679-335-61" boards looks almost exactly the same as "1-679-335-51" ones that worked fine, all the soldering points for the mod are at the same locations.
I'm using ATTiny45 with the latest "master" version of the firmware and soldering instructions. I tried to play around with injection and timing, but it didn't help at all.
Here is debug log from the CD-R (Guilty Gear) loading which ultimately led to CD-Audio player: https://pastebin.com/eb0sip99
And here's the log from the regular game disc (Tekken 3) which boots like it should: https://pastebin.com/NcADhLkx

I'm pretty much stuck at this point, don't know what to do next.
Maybe I should try AttyNee? Does it work with ATTiny45?
Thanks in advance for any help.

ramapcsx2, 2020-02-16:
Is the Guilty Gear log cut short?
It looks like garbage data at first, then some valid subcodes, then it ends.
The initial garbage is fine, as long as it then starts to read valid subcodes.

celcion, 2020-02-17:
Thank you for the fast answer!
No, this is a full log of Guilty Gear loading, PS One actually stops the disc spinning after that. And if you go to the CD Player - it shows two tracks, the second being audio one, obviously. This Guilty Gear disc works just fine on the *-41 and *-51 boards revisions. And I tried to swap the CD drive from those, it didn't help. It's like if PS One also does some other checks before it starts treating the disk like a game disk at all.
And - no, this is not a PM-41 (2) board, the layout is the same as the regular PM-41.
Here's the picture of the board: https://photos.app.goo.gl/PLAXBrBjAA5mwDpi6

ramapcsx2, 2020-02-17:
Ah yeah, okay. 
The CD controller redesign happened with -71, so the WFCK test point is for the regular PM-41 indeed.

So I suppose your -61 expects unlock symbols earlier than all the other boards.
You can test that by reducing the threshold here:
https://github.com/kalymos/PsNee/blob/master/PsNee.ino#L383
Remember to also lower the ceiling accordingly here:
https://github.com/kalymos/PsNee/blob/master/PsNee.ino#L386

(Example: threshold 7, ceiling 5)

celcion, 2020-02-17:
I went all the way to the 0 for the ceiling and it didn't help, even Tekken 3 started to hang on the startup.

Thing is that it's not even trying to spin the disc faster like it should. And whaty I also noticed is that it's not spinning the CD-R discs specifically, any of them (I have about 5 different CD-R disc types). Because when I tried to put just a random factory written CD disc with Windows game - it started to spin it.

And, again, it's not because of the CD drive itself, I specifically tried the drive from the PS One with -41 board which was modded with the same PsNee and ATTiny45 and worked fine - it also doesn't even try to spin up the CD-R disc. But it still detects it as an Audio CD. Like, it slowly spins it and then goes to CD Player in the BIOS. And all the audio tracks reads fine, so it's not just reading problem either.

I don't really know the details, but it looks like it it fails to detect CD-R disc correctly even before modchip have to inject anything.

ramapcsx2, 2020-02-18:
Ah, okay. So yeah, this is a hardware problem.   
I've seen this on an older model PSX once or twice: The CD controller never goes into 2x mode.   
But these machines would also not run official games. They were simply broken.
If your console still runs an official disk, then I assume the CD subsystem is not working up to spec.   
This could be caused by bad capacitors, resistors, even the modchip (if it's soldered on wrong, or not working correctly).

celcion, 2020-02-18:
Yeah, I thought so when I had this with the first -61 board I had. But I just passed the modchip I made with PsNee to the PS One with the -51 board - and it worked right away. And then I had another -61 - and it failed in exactly the same manner. And I also passed the chip to the other board (-11 this time) - and it also worked fine. I never had any problems with the other boards. I soldered them all in pretty much the same way, like this (this one works fine): https://photos.app.goo.gl/GmzEfzRoS7c9zdkG8
I'm not a pro, so of course it could be my shoddy soldering, I won't deny that.

Both of these -61 systems runs official discs, both fails in the very same manner with the modchip. So I could assume they both have some common hardware problem, sure, but I don't have any clue where to look. I think I should also note that they have different CD driver, first one have BA5947FP and the other one have AN8732SB.
Still, you're right, this problem is probably is not related to PsNee whatsoever since it looks like it never comes to PsNee to do anything.

One last question though, if I may - what else could I try to check? Like, try some other version of PsNee and the soldering?
Thank you for your time anyway.

ramapcsx2, 2020-02-19:
Soldering and modchip seem to be fine. I think the CD-R is not being read reliably.
Reasoning: The subcode logs from PsNee look fine, except that it never proceeds to inject.
So I assume that the drive times out and resets reading, then gives up.

celcion, 2020-02-19:
Yeah, but good question is - why? I mean, I tried the CD drive from the PS One where modchip worked without any problems at all. Three of them, in fact. For me it pretty much excludes any possibilities of problems with CD drive itself.
I think that PROBABLY -61 has some BIOS alterations that does some initial disc checking or something. Because it still tries to spin up a regular factory made Windows CD disc with the game (and freeze there, just staying on double spinning speed). This is what makes it really strange, because unmodded PS One doesn't try to spin up the same disc, just showing the standard "Please Insert Playstation Format disc" screen. So it looks like PsNee does its job kicking PS One to detect the regular disc as a game disc, but not for CD-R discs... Which kinda don't make too much sense to me.
I tried all the CD-R types I had, even tried to write on 4x speed with some old IDE CD-RW drive I have - it still doesn't work...
I modded 10 PS Ones so far, one of them being older SCPH-9000 (which I used as a first "victim") - and only those two -61 boards just don't work, all the other works pretty much fine. So I guess we can safely assume that -61 boards probably need some additional research to do in order make PsNee work. I can try to help with that if you're interested and will tell me what to do.

ramapcsx2, 2020-02-19:
Well, as I said, you can pretty much exclude problems with the modchip itself.
The issue is directly with CD-R and that board revision, and the striking difference between CD-R and pressed disks is their lower pit/land reflectivity.

jayjr1105, 2020-07-08:
I realize this is some months back but did you ever figure it out?  I had a similar issue with a NTSC PM-41 board with 1-679-335-61 on it.  Someone on Discord helped me set the fuses correctly and it resolved my issue.  Working fine on my 1-679-335-61 board now.

celcion, 2020-07-08:
> I realize this is some months back but did you ever figure it out? I had a similar issue with a NTSC PM-41 board with 1-679-335-61 on it. Someone on Discord helped me set the fuses correctly and it resolved my issue. Working fine on my 1-679-335-61 board now.

Oh, really? No, I didn't resolve it. Could you please tell me what you did to make it work?

jayjr1105, 2020-07-08:
Keep in mind that I had a SCEA board and this may or may not work for you.  In my case I used an ATTINY85 chip and it wouldn't work at all.  Someone on Discord walked me through setting up the correct "fuses" on the chip and now it works...  I took some notes https://ghostbin.co/paste/4uam3

celcion, 2020-07-08:
Thank you! So, wiring is the same, only fuses should be different?

jayjr1105, 2020-07-08:
Yes, Wiring is identical to any other PM-41 board.

celcion, 2020-07-08:
So, I checked. Sadly, it didn't help. I also noticed that your wiring is slightly different - checked that too. No luck.
It looks like it's some kind of strange SCEI-specific case of -61 boards... I also tried to program it with universal programmer, playing with fuses - no luck either. I guess that I can also try some actual Arduino board instead of ATTiny45, but I doubt it'll change anything...

jayjr1105, 2020-07-09:
My wiring is only slight different because I used different points for ground and vcc.   There are more than one solder point for the other wires as well.  I used this pinout/guide https://quade.co/ps1-modchip-guide/psnee/pm-41/

Too bad it's still not working.  check your solder points with a multimeter in continuity mode.

celcion, 2020-07-09:
I mean, I used a local soldering diagram. This one: https://github.com/kalymos/PsNee/blob/master/PsNee%20modchip%20installation%20guide.pdf
And I altered it for SQCK soldering point. Makes no difference really, but why not.
Anyway, I re-soldered it so many times and on two different boards, I doubt it was a problem with bad soldering - at least once I had to get it right. Also, I successfully modded more than 10 other PS Ones with not -61 boars, including even two -71s (PM-41(2)).
I also tried different ATTiny chips (tried ATTiny13a with different firmware) and other soldering diagrams. Everything failed to work. I can only conclude that it probably have something to do with different BIOS since Japanese PM-41s has region lock that requires disc image patching.

L10N37, 2021-08-29:
I made a new chip not long ago, https://github.com/L10N37/tehUberChip_Another_PSX_Modchip 

No jap console BIOS patching though, although someone released 'psnee V8' about the same time with JAP BIOS patching support.

closed 2023-04-28 by kalymos

ItzSwirlz, 2023-06-19:
Hello, I'm on a PM-41 (2), SCPH-101, NTSC model. Having same issue here and I've tried resoldering

kalymos, 2023-06-23:
I don't know exactly what your problem is, but with my PU41, I had to set
the timer on line 153