kalymos/PsNee #20: PM-41 PAL issue with BIOS patching

greguz, 2020-02-17:
Hi everyone.
I just installed a PsNee (Arduino Pro Mini 3.3v 8MHz no bootloader) inside an SCPH-102 PAL.
PAL games (both backups and original) work fine, but I have an original NTCS game that I'm not able to boot. Already shortened all wires, but no luck. Always show the "dashboard".

From what I understood, the problem may be the [BIOS patch section](https://github.com/kalymos/PsNee/blob/master/PsNee.ino#L175). I suppose the Arduino running at 8MHz may be too slow to patch the BIOS in time. Could it make sense?

Thanks, and nice project!

ramapcsx2, 2020-02-18:
I suspect this is the problem, yes.
You can verify that the patch worked by the video mode the PSone uses for the BIOS.
If it's still PAL, the patch failed. If it's NTSC (the screen is larger than in PAL, and 60Hz instead of 50Hz), then the patch worked and the issue is somewhere else.

Unfortunately, we found some inconsistencies with the BIOS patch. You may want to read about it in the development thread, but you may not find a quick solution:
http://www.psxdev.net/forum/viewtopic.php?f=47&t=1262&start=460

greguz, 2020-02-21:
I just ordered another Pro Mini, a 5V 16MHz version this time. In the next days, I'll do some testing with both the Arduino versions.

Thank you @ramapcsx2

greguz, 2020-03-06:
With the 16MHz version, I was able to boot my NTCS game, and The TV shows a nice "NTCS" label.

I confirm that with the 8MHz the TV is still running on PAL mode, I was not able to make it work.

Even with the 16MHz version, sometimes the game will not boot, like 15% of boots.

ramapcsx2, 2020-03-07:
Okay, thanks for the report :)
I'm not sure how to make this better though, maybe the chip isn't well suited for patching this early after power on.

greguz, 2020-03-07:
Thanks for the support. :)
Hopefully, in the next days, I'll do some more testing with other and more powerful boards.

closed 2020-03-07 by greguz