kalymos/PsNee #25: Support for PS2?

LackofTrack, 2020-12-11:
Hi, I was wondering if this mod chip can be installed in a PS2. I've tracked down all the pins required but I don't know if the PS2 acts different to the PS1, if so is there anyway to make it work?

LackofTrack, 2020-12-16:
Anybody? @ramapcsx2?

ramapcsx2, 2020-12-16:
Oh, this is probably a bad idea.
The theory of operation could apply on a PS2, but due to the various differences, it would basically be whole new project.

I once got a very simple (unlock symbol spamming) modchip working in the PS2, so if you simply use an old 4 wire chip code, that would do it.
The bigger issue is with all the software protections.

LackofTrack, 2020-12-16:
I assume the 4 wire method wouldn't allow for anti modchip games right? That's a bummer, but thanks for answering. Is the PS2 really that different in PS1 mode? I always assumed they would be virtually identical.

ramapcsx2, 2020-12-16:
Yes, a lot of details are different. The hardware is (obviously) PS2 chips, but configured to run in a backward compatible mode.
The tricks to fool the PSX hardware don't apply 1to1.

LackofTrack, 2020-12-19:
I made my own stealth modchip for the PS2 with the help of your code. I found a handy signal that goes low until the PS2 enters PSX mode. From there I just use your code. It works pretty well for the most part.

DraconicNEO, 2021-01-19:
> I made my own stealth modchip for the PS2 with the help of your code. I found a handy signal that is High until the PS2 enters PSX mode (I just spam the unlock symbol while in PS2 mode). From there I just use your code. It works pretty well for the most part. 

Is this for PS1 games on PS2 or is it an actual PS2 modchip? For some reason no one else has had the idea of using an Arduino as a PS2 modchip (Just PS1).

LackofTrack, 2021-01-24:
Sorry, it's just for PS1 games. PS2 modchips seem to be a bit more complicated, perhaps too much for an Arduino.

LMS737, 2021-01-25:
@LackofTrack Hello. I am also interested in getting this to work for PS1 mode inside a PS2. You said that you found a handy signal that indicates the current mode. Do you mind telling us which signal, and where you tapped it from, so it might also help others out? 

Also I wonder what you meant by "works well for the most part". What issues did you encounter, and did you manage to solve it?

DraconicNEO, 2021-01-25:
> 
> 
> Sorry, it's just for PS1 games. PS2 modchips seem to be a bit more complicated, perhaps too much for an Arduino.

Maybe it would work with the Raspberry Pi Pico, or if Not that I'm sure it would defiantly work with a raspberry Pi zero.

LackofTrack, 2021-01-27:
@LMS737 I believe the signal I used was 'GFS', at least that's what the Service Manual calls it. I'm not actually sure what the signal is for but it stays HIGH until the PS2 enters PS1 Mode. So you can use it to determine where you are in the boot process. You'll also have to wire a Reset signal if you want it to work when you reset the console. You can do this by wiring the reset button on the Arduino to the power button on the PS2.

> Also I wonder what you meant by "works well for the most part". What issues did you encounter, and did you manage to solve it?
It only works if you directly go into the game from boot up, meaning you can't go to the menu and then insert your game. I haven't figure out why it won't work when booting from the menu, but for my case it really doesn't matter.

closed 2023-04-28 by kalymos