kalymos/PsNee #28: PU-23 and ATtiny45

Jambul46, 2021-01-02:
can someone please post me wiring diagram for ATtiny45 i can't find it and i don't want to mess it up.

PS MdN:SCPH-9002 PU-23.

Thanks.

Jambul46, 2021-01-02:
and i don't understand this
[image]

kalymos, 2021-01-03:
[image]

Jambul46, 2021-01-03:
yeah i know but i would need something like this 
[image]

But TY.

Jambul46, 2021-01-03:
[image]

Something like this?

Jambul46, 2021-01-03:
[image]
?

kalymos, 2021-01-03:
yes I think that's it, it's been a long time since I modded a ps.

Jambul46, 2021-01-03:
Ok Thanks.

Jambul46, 2021-01-05:
just want to ask i have PAL PSone and i the code it says that bios patch is not supported 
and when i put i game it says me "Please Insert PlayStation CD-ROM"

Jambul46, 2021-01-05:
oh only for PM-41 so  do i have to enable it?

Jambul46, 2021-01-06:
Nice it work's i had to select default region

wyatt8740, 2021-04-06:
You can also just get ground off the "disc lid" switch.

Sorry about the miscellaneous gore on this board. It was a parts board I revitalized.

[image]

closed 2023-04-28 by kalymos