kalymos/PsNee #35: Works great with an ATtiny85 on a PU-8

lagomorph, 2022-12-09:
No issue, just wanted to report success. Serial debug even works at 115200 which surprises me a bit. Every once in a while I'll get the play CD screen with a burned CD but it's pretty rare. Cool stuff!

some-git-user, 2023-01-13:
@lagomorph sorry for taking over your thread, but maybe you could give me some help with attiny85.
I programmed the attiny85 with ardiuno as ISP without a problem.
But after soldering, the PU-23 does not read any game disc. Neither originals nor backups. After reading the disc, the ps1 stuck at "please insert ps1 disc" screen.
But audio CDs working fine.

kalymos, 2023-01-13:
It seems to me of memory it can be the internet clock of 85, which is not perfect.I will see if I find the parameter of the code to modify

lagomorph, 2023-01-13:
Not sure about the PU-23. Are you able to get debug info out of it by defining PSNEEDEBUG? If I connect it up to a console I get something like:

m 0
41000101234900000200257F
41000101235000000200FCFF
010002012351000024669F7F
01000201235200002466FCFF
010002012353000024669F7F
01000301235400054454FCFF
010003012355000544549F7F
01000301235600054454FCFF
010004012357001019509F7F
01000401235800101950FCFF
010004012359001019509F7F
01000501236000154026FCFF
010005012361001540269F7F
01000501236200154026FCFF
!
410100000006000001699F7F
41010000000500000170FCFF
410100000004000001719F7F
41010000000300000172FCFF
410100000002000001739F7F
41010000000100000174FCFF
410100000009000001669F7F
41010000000800000167FCFF
410100000007000001689F7F
41010000000600000169FCFF
410100000005000001709F7F
41010000000400000171FCFF
410100000003000001729F7F
41010000000200000173FCFF
410100000001000001749F7F
41010000000900000166FCFF
410100000008000001679F7F
41010000000700000168FCFF
410100000006000001699F7F
41010000000500000170FCFF
410100000004000001719F7F
41010000000300000172FCFF
410100000002000001739F7F
41010000000100000174FCFF
410100000009000001669F7F
41010000000800000167FCFF
410100000007000001689F7F
41010000000600000169FCFF
...

kalymos, 2023-01-13:
If you want more info on serial output, you can find more info on the original forum. http://www.psxdev.net/forum/viewtopic.php?f=47&t=1262&start=40
Because there frankly it is beyond my poor memory;)

some-git-user, 2023-01-14:
Ok, thank you guys. It's my first programmed chip ever. Will try to enable debugging.

kalymos, 2023-01-14:
Welcome. I never tested the debug on the Attiny, but it seems to me that the output is a little different from the Atmeg

some-git-user, 2023-01-15:
Can confirm it's working on PU-23, but can't really tell why. Tested with Dino Crisis.
Only thing I changed, is the define PSNEEDEBUG flag, but didn't need to debug yet.
Also this time, I choose "Burn Bootloader" before flashing the Attiny85.
Is this even necessary and I forget it in the first place?

kalymos, 2023-01-16:
It's not necessary.
Arduino bootloader operation introduces boot latency, which prevents proper operation of PsNee if you boot the console with a CD in it.

The bootloader is only necessary, when you want to program with the TxTr interface, if you prefer the SIP which does not need a bootloader.

lagomorph, 2023-01-16:
Burning the bootloader also sets the fuses. Maybe an old fuse value was causing issues. 🤷‍♂️

kalymos, 2023-01-16:
I didn't know that it modifies the fuses, it's always good to take as a trick

closed 2023-04-28 by kalymos