kalymos/PsNee #39: Update information on the PSX copy protection

david4599, 2022-12-31:
Hi, I noticed that the "PLAYSTATION 1 SECURITY - HOW IT DOES ITS THING" paragraph is really outdated and should be updated to gather clearer and accurate information, e.g.:

> very little pits and ehm... little not-pits

The "not-pits" are called lands.

> The nifty Sony engineers did not use the pits and stuff to store the security checks for Playstation discs, but went crazy with the rolled up carrier string.

The engineers were not as crazy as it sounds like. They just replaced the 22kHz frequency modulated ATIP (Absolute Time In Pregroove) signal by the 250bps SCEx string (same 22kHz frequency but amplitude modulated).

> the Sony engineers decidedly "fumbled up" the track of sector 4 on a Playstation disc (the track was modulated, in nerd-speak) so that the error correction circuit outputs a recognisable signal, as the laser needs to be corrected to follow the track optimally. This output signal actually is a 250bps serial bitstream

We know the sector 4 has nothing to do with the 250bps SCEx string which is actually located all over the leadin-in (I can confirm from my experiments that it begins at least from the middle of it, sector -6000 and stops just before sector 0). The sector 4 can be seen in any PSX CD image file and contains the license string we see on the boot screen with the 3D PlayStation logo.

kalymos, 2023-01-13:
The part of the text where you propose modifications had been written by the person who had modified the code the most, unfortunately she is no longer really active on this project.
Personally, I was more concerned with trying to gather the documentation, I don't have enough knowledge, in English and in the operation of the system, to modify this part of the text.
 For these different reasons if you feel the courage to send me a version of the improved text I would be grateful.

david4599, 2023-01-17:
I am not the best in English either but here is what would be the modified text, maybe too long but should be accurate enough and hopefully without errors.
And actually, "not-pit" is technically not wrong but the term "land" is always used to designate the opposite of pit.

### The PlayStation 1 security, how does it work?

To protect its precious PlayStation from running unauthorized software, Sony implemented a copy protection mechanism based on a simple ASCII string of text that is read from a part of an original PlayStation disc that cannot be reproduced by an ordinary PC CD burner.

Basically, a CD is made from a really long spiral of pits and lands that represent the data stored on the disc. Inside the Optical Pickup Unit (OPU), a laser diode emits infrared light that goes through lenses and mirrors, hits the disc, bounces back and shines on a sensor made from at least 4 photodiodes. These photodiodes receive different amounts of light depending on pits and lands' reflectivity allowing to recover the data but they are also used to perform self-adjustment to keep the track aligned and in focus.
This is done by comparing photodiodes outputs in a way to create the Tracking Error (TE) and Focus Error (FE) signals and then send them to the CD controller to move the lens head horizontally for the tracking and vertically for the focus.

On a CD-ROM (read-only disc pressed inside a CD factory), the spiral is a simple rolled up line. However, on a blank CD-R, the spiral i.e. the pregroove is wobbling on all over the disc and the CD controller follows it. This wobble picked up by the TE signal actually carries data known as the Absolute Time In Pregroove (ATIP) and is frequency-modulated at 22kHz (1x read speed). The ATIP contains information about the disc characteristics but is useful only for burning.

Now, what is the relation between these explanations and the PlayStation security?

As PlayStation discs are read-only mediums, they do not need the ATIP. Sony engineers surely thought:

> Why not reuse the wobble that burners can't reproduce (in theory) to encode some sort of information that will provide copy protection?

And that's what they did! In the lead-in area of a PlayStation disc (negative sectors), the spiral is wobbling at the same 22kHz frequency like a CD-R but instead of storing the ATIP, they stored a 250bps amplitude-modulated serial bitstream (with 1 start bit and 2 stop bits) that repeats until sector 0 (end of lead-in).

The important thing to understand is the bitstream is not part of the game data which is encoded using pits and lands' length. Instead, it is made by slightly shifting the pits and lands' position. The self-adjusting system simply follows this wobbling track creating a modulated TE signal that needs to be demodulated using a 22kHz filter. Then, it goes to the PlayStation sub-CPU for decoding.

The bitstream in plain ASCII says "SCEA" (Sony Computer Entertainment of America), "SCEE" (Sony Computer Entertainment of Europe) or "SCEI" (Sony Computer Entertainment Inc. (i.e. Japan)), depending on the region of the disc.
The security thus functions not only as copy protection, but also as region protection.

The text string from the disc is compared with the one that is embedded in the PlayStation sub-CPU. If these text strings are the same, the disc is interpreted to be authentic and from the correct region. Bingo!

kalymos, 2023-01-17:
great, I'll take a look to get this straightened out when I get back from vacation ;)

kalymos, 2023-02-05:
I went home, reread the text, and made the changes.
Thank you for your contribution.

david4599, 2023-02-06:
You're welcome!

closed 2023-02-06 by david4599