kalymos/PsNee #40: RPi Pico support?

xslendix, 2023-01-28:
Those are some really cheap microcontrollers, and are easy to come by. I think having support for those would be quite cool.

VoxelTek, 2023-04-27:
I agree, and I'm trying to port the code to it now. I literally have not tested this code and am not an Arduino or PS1 security guru by any means, but I figure I might as well give it a shot.

[https://github.com/VoxelTek/PicoNee](https://github.com/VoxelTek/PicoNee)

kalymos, 2023-04-28:
Good initiative, here I will keep the code for attiny and attmega.
Tell me when the code will be functional I will put a link.

remoitrix, 2023-06-14:
Hi!
I compiled the code and show this errors;

In file included from C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\variants\SEEED_XIAO_RP2040/pinmode_arduino.h:30:0,
                 from C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/Arduino.h:26,
                 from C:\Users\thesp\AppData\Local\Temp\arduino\sketches\05397FCBE827B03C5F7D492FE9E31C1C\sketch\sketch_jun14a.ino.cpp:1:
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino: In function 'void inject_SCEX(char)':
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:103:24: error: 'PINB' was not declared in this scope
   #define GATEWFCKPORT PINB   // MCU port for the gate input (used for WFCK)
                        ^
C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/api/Common.h:67:32: note: in definition of macro 'bitClear'
 #define bitClear(value, bit) ((value) &= ~(1UL << (bit)))
                                ^~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:174:16: note: in expansion of macro 'GATEWFCKPORT'
       bitClear(GATEWFCKPORT, DATABIT); // data low
                ^~~~~~~~~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:103:24: note: suggested alternative: 'SING'
   #define GATEWFCKPORT PINB   // MCU port for the gate input (used for WFCK)
                        ^
C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/api/Common.h:67:32: note: in definition of macro 'bitClear'
 #define bitClear(value, bit) ((value) &= ~(1UL << (bit)))
                                ^~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:174:16: note: in expansion of macro 'GATEWFCKPORT'
       bitClear(GATEWFCKPORT, DATABIT); // data low
                ^~~~~~~~~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:103:24: error: 'PINB' was not declared in this scope
   #define GATEWFCKPORT PINB   // MCU port for the gate input (used for WFCK)
                        ^
C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/api/Common.h:65:32: note: in definition of macro 'bitRead'
 #define bitRead(value, bit) (((value) >> (bit)) & 0x01)
                                ^~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:183:38: note: in expansion of macro 'GATEWFCKPORT'
           bool wfck_sample = bitRead(GATEWFCKPORT, GATEWFCKBIT);
                                      ^~~~~~~~~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:103:24: note: suggested alternative: 'SING'
   #define GATEWFCKPORT PINB   // MCU port for the gate input (used for WFCK)
                        ^
C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/api/Common.h:65:32: note: in definition of macro 'bitRead'
 #define bitRead(value, bit) (((value) >> (bit)) & 0x01)
                                ^~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:183:38: note: in expansion of macro 'GATEWFCKPORT'
           bool wfck_sample = bitRead(GATEWFCKPORT, GATEWFCKBIT);
                                      ^~~~~~~~~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:104:20: error: 'PORTB' was not declared in this scope
   #define DATAPORT PORTB      // MCU port for the gate input (used for WFCK)
                    ^
C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/api/Common.h:66:30: note: in definition of macro 'bitSet'
 #define bitSet(value, bit) ((value) |= (1UL << (bit)))
                              ^~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:184:11: note: in expansion of macro 'bitWrite'
           bitWrite(DATAPORT, DATABIT, wfck_sample); // output wfck signal on data pin
           ^~~~~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:184:20: note: in expansion of macro 'DATAPORT'
           bitWrite(DATAPORT, DATABIT, wfck_sample); // output wfck signal on data pin
                    ^~~~~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:103:24: error: 'PINB' was not declared in this scope
   #define GATEWFCKPORT PINB   // MCU port for the gate input (used for WFCK)
                        ^
C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/api/Common.h:67:32: note: in definition of macro 'bitClear'
 #define bitClear(value, bit) ((value) &= ~(1UL << (bit)))
                                ^~~~~
[... cut ...]
C:\Users\thesp\AppData\Local\Arduino15\packages\Seeeduino\hardware\rp2040\2.7.2\cores\arduino/api/Common.h:65:32: note: in definition of macro 'bitRead'
 #define bitRead(value, bit) (((value) >> (bit)) & 0x01)
                                ^~~~~
C:\Users\thesp\Desktop\mods\psone hard-softmod\Mod cambio Bios\PsNee-7.01\PsNee-7.01\test\sketch_jun14a\sketch_jun14a.ino:341:22: note: in expansion of macro 'SUBQPORT'
     sample = bitRead(SUBQPORT, SUBQBIT);
                      ^~~~~~~~

exit status 1

Compilation error: 'PINB' was not declared in this scope

dazzaXx, 2024-03-18:
Is there any update on the ported pi version? I have a bunch of Picos and RP2040 Zeros that I could use with it.

remoitrix, 2024-03-18:
Nothing yet. I think that is a forgiven project.

VoxelTek, 2024-03-19:
Unfortunately, I didn't have the necessary hardware in order to test my code and such. I was unable to even get the original PSNee working, and as such testing and debugging my code is somewhat out of my reach.

Apologies to everyone involved, feel free to take over the project if you'd like to help continue my work.

dazzaXx, 2024-03-19:
No worries, I did try to also compile [kalymos's version of PicoNee](https://github.com/kalymos/PicoNee) and noticed some defines were apart of some sort of [port manipulation](https://docs.arduino.cc/retired/hacking/software/PortManipulation) being used, and because we try to compile for rp2040 boards, the code for them doesnt include this I assume, thats why it errors on PINB, PORTB, etc.

I'm no coder, but this is as far as I could get with figuring out why it doesnt like to compile.

closed 2024-09-11 by kalymos