kalymos/PsNee #42: Should the RX and TX LED's be permamently on in the Pro Micro?

FaxMac, 2023-04-25:
Hello, After I installed the code via SPI to the Pro Micro that I am going to use for this project, I noticed that when powering on the board, the RX and TX LED's are constantly on, But while checking the code, I noticed that there is no logic or line that turns on those LED's.

To other users that used a Pro Micro in this project, Is this normal behavior?

FaxMac, 2023-04-27:
> […](#)

That's the thing, I remember specifically not uncommenting the #include debug

kalymos, 2023-04-27:
I took a look in the code, between line 250 and 256, we see that the serial
port is open under all conditions, and I'm not sure the right way to change
that.
Perhaps by commenting out lines 253 to 256, i tested the compilation it
passes, but it will be necessary to test if the mode is still functional
after that.

hope this is useful

closed 2024-09-11 by kalymos