kalymos/PsNee #43: PM-41 NTSC J

some-git-user, 2023-07-22:
ATtiny85 does not load originals/backups from PAL region, but japanese originals/backups running fine.
I can repeatedly hear the disc spinning up/slowing down until I land in the memorycard/cd player menu. 
While in the menu, the disc is still spinning up/slowing down (I don't see the "insert PS disc" screen).

wyatt8740, 2023-07-24:
Mid/Late model Japanese systems have to have the discs patched to claim to be NTSC-J or have their entire BIOS chips replaced, because there's a secondary check in the BIOS code that the modchip (which manipulates the mechacon) cannot bypass (yet, at least). The only PAL PSX that does this is the PM-41(2), and no north american units do the check.

So it's easiest to patch your game discs so that the BIOS thinks they're NTSC-J region. But it's not really something within the scope of this modchip to be able to fix (unless someone makes a hack like the one for the PAL PM-41(2) - which the ATTiny85 might not have enough pins to do anyway).

some-git-user, 2023-07-25:
Thanks for the explanation. I think this is very important information and should be written down in the Readme. Up until now I thought only PAL PM-41 boards needed a BIOS patch to run region free.

kalymos, 2024-09-11:
By the way if you are still interested, a new version is in the works that supports all Japanese models.

https://github.com/kalymos/PSNee_V8/tree/test2

closed 2024-09-11 by kalymos