kalymos/PsNee #45: Question  about running mod

arminsalcin, 2023-09-10:
Hello,

Thanks for library, i used attiny85 and programmer avrisp i succesfully burned chip (with no bootloader) and 8Mhz internal clock but after installation i cant run original PAL game, my system is SCPH-101 with PM-41 (2) board. 

Question is : 
- Can we still run original games after chip?
- Will PAL games work after the chip on NTSC console?
- Is burning fuses on attiny really necessary before uploading code via arduino IDE? (I didnt do this part)

kalymos, 2023-09-10:
Hello,

Overall the answer is yes to all three questions, for the fuses if you
install the Arduino bootloader, and then you burn the program again the
fuses remain.

But normally on SCP-101, you have to run the bios patch, which is not
really possible on ATtiny85.

Sorry

arminsalcin, 2023-09-10:
Thanks for the quick answer, i got confused by the bios patch [here](https://quade.co/ps1-modchip-guide/psnee/pm-41-2/) It says only if you have PAL SCP-101 to do bios patch, I will try with Arduino pro mini definitely to move out of probably mistakes with burning fuses.

closed 2024-09-11 by kalymos