kalymos/PsNee #47: PU-8 playstation , using arduino mini pro 8 mhz, no disk detected with psnee

interfan, 2023-12-15:
PU-8 playstation , using arduino mini pro 8 mhz, no disk detected with psnee.

I have ps1 PU-8.

I used arduino mini pro 8mhz model.
PSNEE uploaded and the line for the mini uncommented.
All wires soldered as per diagram.
After the modchip is installed, the disks are spinning but i see the menu for the audio disk or memory card.
if i disconnect the chip the ps1 is working again with original disk.

What could be the issue?

kalymos, 2023-12-16:
hello a user pointed out to me that I had made an error in the wiring
diagram for the Pu-8 which I have just updated. check that you are using
the latest version of the PDF. If it persists it could be due to timing
issues.

interfan, 2023-12-17:
My wiring is correct as per latest diagram.
I tried increasing the attempts to (inject symbols now. 2 x 3 runs seems optimal to cover all boards) from 2 to 3.
#define ARDUINO_328_BOARD
I have also tried #define APPLY_PSONE_PAL_BIOS_PATCH
I have also tried to set delay_between_injections o 72

Im tottaly blocked.....with this one.

Disk is spinning (original disk) , but i am sent to the screen for audio disk and memmory card.

How i can debug what is going on?
Any one with the same board that can help on this?

closed 2024-01-17 by interfan

ba114, 2025-03-26:
What was the resolution to this? Im having the same problem with my PU-8

kalymos, 2025-03-26:
If you are using an 8MHz ATmega, the current code is more compatible.
maybe V7 version can work.
But if it is not that, to help you I will need the console number, we already have the motherboard number, I also need to know the number of your MCU, your compilation options, what is the compilation and injection chain that you used, and also a photo of the soldering.

ba114, 2025-03-26:
Fixed my issue. I was using AVRDUDESS GUI to flash the compiled HEX but i hadnt set the fuses correctly for ATTiny45. I was using v8.5 ino which doesnt actually specifically state the fuse values for the tiny (unless i missed it).

kalymos, 2025-03-26:
I'm very glad it's going to work for you, all the necessary information is in the Wiki, it's one of the big changes I made in this version. 
It allowed me to better classify the information and include a lot of details.