kalymos/PsNee #53: Does PSNee need to be flashed via SPI for a PU-7 board?

dazzaXx, 2024-03-21:
Hi, just wondering if SPI flashing is needed for a PU-7 board, I have an Arduino Nano on the way and noticed some mentions of SPI flashing the comments of the .ino code file as well as around the web, I presume that is for the "PM-41 patching" portion but I just wanted to double check.

Thanks!

kalymos, 2024-03-21:
The best is always to flash in isp, this eliminates the boot code.

The easiest way to do this is to transform an Arduino into an ISP
programmer, with the Arduino as ISP sketch

dazzaXx, 2024-03-21:
Do you have any recommendations for SPI Programmer sketches?

kalymos, 2024-03-22:
The one that is available in the Arduino ;)

https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP/

dazzaXx, 2024-03-22:
Thank you, I'm a bit of a noob when it comes to arduino and ive never used an ISP flasher before, i've only used the upload button in the IDE xD

dazzaXx, 2024-03-23:
Hi, i've just finished making a ISP flasher with my spare mega2560 I had lying around and tested flashing another arduino with it and it finally works! _(I'm not a coder at all so it took me HOURS to figure it out)_

So for now i'll close this issue so it's not left open. Now it's ready for when my nano arrives for PsNee!

Thanks for the help mate, I really do appreciate the help and tips. 👍

closed 2024-03-23 by dazzaXx

kalymos, 2024-03-23:
two little extra tips, using the ISP you can install the bootloader (in
Arduino choose the right card, and in the options installed the bootloader)
installing the bootloader with Arduino modifies part of the registers
(often called fuse ) which sometimes allows the Attiny to work correctly

dazzaXx, 2024-03-23:
Is that needed for a Nano? I know when you flash with ISP it removes the bootloader for faster boot times which is needed for this use of PsNee, is the Burn Bootloader option for say if I wanted to upload normally to it again?

kalymos, 2024-03-23:
Normally it's not necessary for a Nano, and for the rest that's it