kalymos/PsNee #54: PU-7 all game discs being read as Audio CD's

dazzaXx, 2024-03-25:
Basically the title, i've installed PsNee with my arduino, but now, even with legit discs, it always kicks me to the CD player.

It's most likely me thats done something wrong, but what could it have been?

dazzaXx, 2024-03-25:
It seems it may have to do with the voltage the nano is recieving, i disconnected the 3.3v from the nano and plugged in a usb lead and it seemed to read them fine. I dont suppose there is any 5v pads i could test out?

kalymos, 2024-03-26:
For this project we always power the Arduino with 3.3v (we are at the limit
of the recommended voltage for a 16 mh clock), but this allows us to be
sure that the signal sent to the ps does not destroy it.
The first thing to check is the wiring.

dazzaXx, 2024-03-26:
Well on my edits, the ps1 doesn't even read anything anymore, once the lid sensor is put down and released, the ps just spins forever until its reset, and the nano doesn't even output pulses to show when its injecting anymore. 

I enabled debug and did see it was attempting to inject, but now nothing even happens.

dazzaXx, 2024-03-26:
It wouldn't suprise me if I've fried something on the board.

kalymos, 2024-03-26:
I won't tell you that you have to fear the worst.
If you read the output of the Arduino through the USB, there is no longer a
conflict between the USB power supply at 5v and the internal voltage of the
Playstation....

already trying to disconnect the VCC from the Arduino, and also unplug the
USB, this should prevent the Arduino from turning on

dazzaXx, 2024-03-26:
Unfortunately I have already disconnected vcc and ground from the Arduino and even unplugged the usb. And the results were still the same.

kalymos, 2024-03-26:
Unfortunately I think she is dead R.I.P 😔

dazzaXx, 2024-03-26:
No worries, I figured that was the case. Thank you for the help.

kalymos, 2024-03-26:
It may be an input output of the Arduino which is broken.
Personally I will install the bootloader and I will test the inputs and
outputs of the Arduino

dazzaXx, 2024-03-26:
It could also be the fact that Ive made the wires too long, as I didnt read properly that theyre meant to be as short as possible.

dazzaXx, 2024-03-26:
Hi, I've finally fixed it and its working fully now. I shortened the wires as well as resoldered the ground on the nano as it turned out that it was loose.

Thanks again for the help and tips!

closed 2024-03-26 by dazzaXx

kalymos, 2024-03-26:
Glad to know all is well.
 good DIY