kalymos/PsNee #58: SCPH 100 question

kschewe, 2024-11-16:
I have the JP version of the ps1.  I am using the Arduino Nano. 
 In you code i #define ARDUINO_328_BOARD
In the instructions it does not mention to wire to bios chip, it only mentioned the PAL version.  Is this correct?
I was not able to get it to work at the moment.  My other one i install mod chip and change the bios on the JP version

kalymos, 2024-11-16:
hi. if it is indeed a 100 then a Japanese one. this version of the mode does not work with it, look instead at version 8.5, which works with it, and which has a new documentation.

https://github.com/kalymos/PSNee_V8/tree/test2

kschewe, 2024-11-16:
Thanks for the response
I read the Doc on this one for the pu-41.  i will try and get back to you.
Thanks

https://github.com/kalymos/PSNee_V8/tree/test2

kschewe, 2024-11-17:
psnee v8.5 works on schp-100 JP version.  The 2 additional points needed be soldered tot he bios chip DX and AX.  8 wires in total.  I used the Arduino nano.  Using Arduino ide.  All i did was define atmega328 and define scph-100.  Programmed by usb com port.  Nothing else was required.

kalymos, 2024-11-17:
Does it work with region-free or burned games? If so, it's good.
but if you programmed it without ISP, the bootloader is still on the card, and it can cause a delay that prevents the bios patch.

kschewe, 2024-11-17:
I just tested and it worked on legit NA, JP and burnt game.  So far it loaded up all games with no issues.  I intend to install on a schp 1000 model soon.  I could try again with the other method since it seems to have no issues programming the nano when installed into ps1.
 I did not try the ISP way since it was first time using this software and Arduino.

kalymos, 2024-11-17:
I'm happy for you.
P.S. Sometimes using ISP can be a bit of a pain, to test I use a blink sketch.

ba114, 2025-02-15:
> hi. if it is indeed a 100 then a Japanese one. this version of the mode does not work with it, look instead at version 8.5, which works with it, and which has a new documentation.
> 
> https://github.com/kalymos/PSNee_V8/tree/test2

Is the latest version you published here now fully compatible with SCPH-100?

kalymos, 2025-02-15:
Hello
Normally the mode is compatible with your console, taking into account that the Attiny are not compatible with the bios patches that are necessary for Japanese consoles, you are forced to use the ATmega.
Unfortunately as I do not have this console personally, and I do not have any feedback yet, I have not yet put it in the table as being guaranteed.
But even without the bios patch not being functional, you will still be able to make burned Japanese games work for me.

If you do I would be happy to have your feedback so that I can put this console table to test.

ba114, 2025-02-15:
> Hello Normally the mode is compatible with your console. For the BIOS patch to work you need to program with the ISP. I made a tutorial to be able to program with the ISP, https://github.com/kalymos/PsNee/wiki/Personal-recipe-to-make-a-good-PSNee Unfortunately as I do not have this console personally, and I do not have any feedback yet, I have not yet put it in the table as being guaranteed.
> 
> If you do I would be happy to have your feedback so that I can put this console table.

Have just programmed a pro micro using my UNO R3. Will look into wiring it up tomorrow and see what happens

ba114, 2025-02-16:
installed today. im having trouble with the pinouts.

The wiring diagram here (https://github.com/kalymos/PsNee/wiki/PM%E2%80%9041-diagram) shows pins AX and DX labelled, which i have run to pins 2 and 4 respectively on my pro micro. Where is the AY pin on the motherboard and where doi wire reset to?

With it wired as is, i get "please insert a PlayStation format disc" regardless of what disc is installed. I dont have a genuine NTSC-J disc to test with, but i assumed my PAL disc would show a region error rather than saying to insert a disc.

kalymos, 2025-02-16:
Your pike connection is correct, on the PM_41 we do not use the AY pin.
The bios patch seems suitable otherwise it would cause a black screen. (but it is not 100% guaranteed)
Was there a game in Japanese version in the burned CDs that you used, it also does not work it would seem to indicate a problem with the injection of the region code.
And using a normal music CD, can give you a unit of the state of the player.
You can use the LED run option, with this letter will light up just after the console starts to signal the injection of the bios patch, and a second time when reading the region zone. This will already give you an indication that the chip mode detects the moments when it must inject the code.
Feel free to use a multimeter in continuity mode to triple check if the wiring is correct, I always do this because I always get it wrong ;)

ba114, 2025-02-16:
> Your pike connection is correct, on the PM_41 we do not use the AY pin. The bios patch seems suitable otherwise it would cause a black screen. (but it is not 100% guaranteed) Was there a game in [...]

Tried with Japanese burnt cd and get the same error.
Wiring was double checked with multimeter for continuity and all good there.

Can hear the laser on the drive trying to adjust to read the discs but end result is the same on all discs I try.

The red led on the arduino is always on, I don't seen any other activity. I'll enable the led run code and try again to see what happens.

kschewe, 2025-02-16:
I was successful on mine.  I tested JP NA legit and burnt game.  I am using a different chip though and that might be the difference.  I did not get around to try on the 1000 JP model i have yet.  I was going to investigate burning it to a single chip instead of these little boards for more room.

kalymos, 2025-02-16:
It's good that you succeeded, I could just know the detail, what was the Arduino board, the console number, and the motherboard number. 

This is to complete the success table

ba114, 2025-02-16:
should i be usings v8 or v8.5.2?

kalymos, 2025-02-16:
v8.5.2

ba114, 2025-02-16:
> v8.5.2

yep thats what im using...

kschewe, 2025-02-16:
here are are few pics.  

[image]
[image]
[image]

ba114, 2025-02-16:
Ok all working now. exported the compiled HEX and reflashed using avrdudess setting the fuses in there.
Tested with Jap and NA burned discs and original PAL discs. All loaded fine.

kalymos, 2025-02-17:
Good that it works, but ultimately it was an injection problem.

kalymos, 2025-02-18:
one last thing i could have the command line you used for arvdude

ba114, 2025-02-18:
> one last thing i could have the command line you used for arvdude

i used avrdudess gui for it

ba114, 2025-02-26:
> one last thing i could have the command line you used for arvdude

avrdude.exe -c avrisp -p m32u4 -P COM7 -b 19200 -U flash:w:"C:\Users\xxxxx\OneDrive\Documents\Arduino\Projects\PSNee_V8.5\build\PSNeecore.avr.32u4\PSNee_V8.5.ino.hex":a -U lfuse:w:0xFF:m -U hfuse:w:0xDF:m -U efuse:w:0xFF:m

closed 2025-03-19 by kalymos