kalymos/PsNee #60: PM-41 PAL problem

impy101, 2024-12-08:
I've spent the past 5 hours trying to get PSNee to work on my PAL PM-41 but nothing works. I've used Arduino Uno as a programmer and a FT232RL to program Arduino Nano's and Mini-Pro's and neither work. I definitely have everything wired correctly and I checked with a multimeter that everything is reading right. But the PSOne still refuses to read burned discs that my older PSX does read.

kalymos, 2024-12-08:
At first glance like this it's hard to tell. Did you compile and connect
correctly with the bios patch.

by the way I give you the link to the new version, with completely revised
documentation
https://github.com/kalymos/PSNee_V8/tree/test2

impy101, 2024-12-08:
> At first glance like this it's hard to tell. Did you compile and connect correctly with the bios patch. by the way I give you the link to the new version, with completely revised documentation [...]

Same thing I'm afraid. I definitely compiled it properly and there's nothing wrong with my wiring. 

My FT232RL to Pro-mini wiring: VCC - VCC, GND - GND, TX - RX, RX - TX, DTR - DTR. Board settings in arduino IDE are correct and for programmer I used Arduino as ISP and AVRISP MKII. Still nothing, all I get is my PSOne sees the discs as some kind of media with minutes etc.
 

[image]

kalymos, 2024-12-08:
you tested with a genuine disc from another region?

impy101, 2024-12-08:
> you tested with a genuine disc from another region? I'm not 100% sure, but it looks like the bios patch isn't working. Le dim. 8 déc. 2024 à 11:26, impy101 ***@***.***> a écrit :
> […](#)

I don't have a genuine disc from another region unfortunately. However, I noticed in the guide that I should be uploading the code with sketch -> upload using programmer. I tried that several times, but I just get: avrdude: stk500_recv(): programmer is not responding

However, when I upload "normally" using Arduino ISP it does succesfully upload.

kalymos, 2024-12-08:
The recipe I use is really specific, and must be followed step by step.
But if your ISP method works, use it.
I'm not 100% sure, but it looks like the bios patch isn't working.

P.S. Little extra tip, used the LED_RUN option, it told you if the region code is sent

kalymos, 2024-12-08:
P.P.S Injection with Arduino ISP is always a bit tricky, personally I always test first with the blink sketch ;)