kalymos/PsNee #63: Black screen on SCPH-1000 with 1.1j BIOS

notsocafe, 2025-02-09:
(I am aware that GitHub issues are not the best way to discuss this, but I've yet to find any other trustworthy online exchange about this)

I tried to build a PsNee for my SCPH-1000 out of an Arduino Nano clone (Whadda WPB102), but the console no longer boots when BIOS patching is plugged in.

Arduino has defines Atmega328_168 & SCPH_3000 uncommented, flashing is done through ArduinoISP via a proxy Uno.

When everything besides BIOS lines are plugged in, the console works fine, but does not read any other region disc.
When BIOS lines are plugged in the Arduino, TV reports AV/RGB signal going through, but it's only black screen.

All connections are OK through continuity testing, although non-BIOS lines are using longer cables.

Is there a bug, or something to investigate?

(Available for discussion in French if needed.)

kalymos, 2025-02-09:
I feel less alone, I just received a SCPH-1000 and I have the same problem,
today.
indeed like you I think that it comes from a problem with the injection of
the bios patch, the problem is that the Russians who made the bios patches
did not comment on their work at all, I will have to recheck the wiring.
unfortunately if it is the code that is messed up I can't do much.
a priori if you have already installed some that worked, it is because your
cables must not be too long. on all the ones that I have I have the cables
that come out of the box with harnesses, to be able to test the different
chipsets.
je parle aussi français, mais pour la réponse cela donnait en anglais si
des gens avaient le même souci

notsocafe, 2025-02-16:
I can confirm that the modchip works with the correct BIOS diagram.
Thank you for the hard work.
Maybe we'd need to update the comments in the .ino to match the IC102 pinout.

closed 2025-02-16 by notsocafe

kalymos, 2025-02-16:
the information in the .ino is correct it was the diagram that was wrong

just one more thing, could you tell me again on which machine you managed to install the mode, giving me the console number, the motherboard number, and which Arduino board you used.

ba114, 2025-03-26:
I just installed on SCPH-1000 (40 pin bios) as well. Anduino Pro Mini (5v16MHz),  Uncommented #define SCPH_1000.

Boots original Jap and Backup discs but doesnt boot original or backup games from other regions.
Followed the BIOS pin pictures from the wiki and the pinout diagram for pro mini.

kalymos, 2025-03-26:
What method did you use to inject the code into the pro mini?

ba114, 2025-03-26:
> What method did you use to inject the code into the pro mini?

Exported compiled code and flashed using USBasp in AVRDUDESS. Fuses were set per wiki.

kalymos, 2025-03-26:
I'm not really familiar with this injection method, but it should work so you can play Japanese copy discs.
To be sure of the bios version, the bios dumper disk will give you the references that correspond to the one in the code.
https://github.com/kalymos/PsNee/wiki/Just-what-you-need-to-know-to-know-your-BIOS-with-BIOS-Dumper-2.6.
After that the last thing on my mind is a problem with the connections.

ba114, 2025-03-26:
> I'm not really familiar with this injection method, but it should work so you can play Japanese copy discs. To be sure of the bios version, the bios dumper disk will give you the references that [...]

Yes it works with NTSC-J copy discs. Just doesnt work with NTSC-U or PAL (original or copy)