kalymos/PsNee #66: Attiny45 and UNO as ISP -- Upload error:  avrdude: invalid byte value ({upload.low_fuses}) specified for immediate mode

ivancosta, 2025-02-13:
I'm programming an Attiny45 through an Arduino Uno as ISP and ARDUINO IDE and I get this error when trying to upload using the programmer: 

In the Wiki and in the instructions PDF it's mentioned that settings have to be made in the fuses, but NO fuses options are visible in the IDE.

I'm using the method of copying the PSNeecore folder at the arduino15/packages/ directory.

What am I doing wrong?

Console log:

```
Sketch uses 844 bytes (20%) of program storage space. Maximum is 4096 bytes.
Global variables use 30 bytes (11%) of dynamic memory, leaving 226 bytes for local variables. Maximum is 256 bytes.
avrdude: invalid byte value ({upload.low_fuses}) specified for immediate mode
avrdude: read from file '{upload.low_fuses}' failed
Failed programming: uploading error: exit status 1
```

kalymos, 2025-02-13:
Thanks for the feedback
I think I found where the bug is, I'll have to check tomorrow.

And it's normal that you can't find the fuse option for the Attiny, since we can't use them with the bios patch.

kalymos, 2025-02-13:
In the meantime you can use one of the standard Attiny libraries, it won't change much in the recipe

ivancosta, 2025-02-13:
When you say "libraries" you mean a board manager, like https://github.com/damellis/attiny ?

kalymos, 2025-02-13:
If you are here, I have fixed the bug, the problem is that I don't have an ATTiny at hand at the moment.
Could you test it for me?
If yes I have uploaded a new PSNeecore file, just replace it in your folder.

ivancosta, 2025-02-15:
I've tested and the bug is fixed

closed 2025-02-15 by ivancosta

kalymos, 2025-02-15:
Thank you very much for your reply.

Can I still ask you something?
What is the console number where you installed the mod, the motherboard number, and the ATtiny you used?

ivancosta, 2025-02-15:
I'm trying to mod a 7001, motherboard PU-20 with Attiny45.

Flashed the Attiny, soldered and it's not functioning. 

In the .ino sketch I've **uncommented** these 2 lines only:
` #define ATtiny85_45_25`
and
`#define SCPH_xxx1`

The wires I'm using are not super short, they are like 7cm. 
I am missing something?

kalymos, 2025-02-15:
It's not the length of the cables, on my test machines more than 20 cm.
If the programming by ISP went well, the first thing I suspect even at home is to have soldered the cables in the right order and in the right place, you can send me pictures of the soldering on the motherboard.

I personally did tests on the 7002 pu-20 version.

ivancosta, 2025-02-15:
Here are my soldering points

This board came to me having a modchip already installed and it was not working either.

But the console runs backup disks normally with softmod FreePsxBoot/Unirom

[image]
[image]

kalymos, 2025-02-15:
At a glance on the welds I see one or two small problems, the part where you removed the cable sheath is a little long, generally I arrange it so that it is between 1 and 2 mm.

And the soldering of the SUBQ and the wfck are not very beautiful.

Personally to have good welds first on the point where I want to mess up I add a little solder, I expose the cable, I put a little solder on the end of the cable, and if necessary I shorten it, then I follow the cable and the point.

And finally I check if I have no false contact with a multimeter.

One last thing, if it welds badly, add flux.

ivancosta, 2025-02-15:
Yeah, not proud of my soldering on that. 
I redid those and it still do not work.

Monday I will try flashing the Uno, to tell if make any difference, maybe it is an issue with my board / OR the Attiny.  
I have an old analog osciloscope at hand, too, to poke around.

PS: a have flashed a bootloader to the Attiny to test with the blink sketch, and it was ok. 
Noobie question: bootloader flashed prior to PSNee, ok? not ok?

Thanks for the support!

kalymos, 2025-02-16:
It's good that you're trying to check that if you injected a sketch on the target, but at this stage a simple multimeter in continuity mode is sufficient to check if the welds are good, and that all the cables are in the right order.

ivancosta, 2025-02-17:
I've soldered a socket and tested with an Arduino Uno R3, it works fine. Put the Attiny back on, and it no longer works.

kalymos, 2025-02-18:
apparently there is a problem with injecting the code into the ATtiny, someone managed to solve the problem next door, using arvdude directly. I'll ask him for his command line

kalymos, 2025-03-14:
I'm glad it worked for you. I was a little worried I'd bricked your ATtiny with the wrong fuse settings.
If you've checked out the new wiki and its tutorial, what do you think?
Have a nice day!

ivancosta, 2025-03-14:
Well, I have bought new Attinys, the ones (2) I was using don't accept any programming anymore. I think they are bricked, but don't worry with that, your new updates came right in time when the new chips have arrived! Thanks again!   
I

kalymos, 2025-03-14:
There are ways to recover, you can put a 16 MHz clock on the right pins (I made a mistake in setting the fuses and I tell them an external clock), you can reprogram them again via ISP, to change the fuses. Or use a more advanced programmer like the Chinese t48.

ivancosta, 2025-03-14:
I'll take a look at those later, with a 16Mhz crystal. Thanks for suggesting that.
The wiki is becoming more helpful!

I'm not an english native speaker, so pardon my past, present and future text errors :)  

I noticed some errors in the text and suggest the following corrections at the top of the wiki page. 

"MUC" → "MCU" (Assuming you meant "Microcontroller Unit").
"current supporters" → "currently supported" 
"Not compatible with BIOS patch" → "Not compatible with the BIOS patch" 

and so on...

kalymos, 2025-03-14:
Don't worry, let's just say English is my third language ;)
I looked at the schedules you mentioned.