kalymos/PsNee #68: Difficulty installing on the Japanese SCPH 1000

godmanpicoli, 2025-02-19:
I have already checked the installation of the wires on the Arduino Nano and on the PS1 board. I cannot start games from another region without the PS screen changing colors. Some games freeze on the PS screen. However, I can run the CD-R with the PSX BIOS Dumper.
Japanese games work normally.

Arduino Nano Fuses: H: DF, L: EE, E: FF
PS1 SCPH_1000 PU7 BIOS ver. 1.1 CRC 3539DEF6 checked with PSX Bios Dumper

[image]
[image]
[image]

kalymos, 2025-02-19:
for european games it's normal that it doesn't work very well, the internal clock of japanese ps1 doesn't allow it. 
there is a mode to change the clock, at the end of the first page of the wiki you have a link to this world

godmanpicoli, 2025-02-19:
I forgot to mention that I used American games, even with American games do I need to change the PS1's internal clock?

kalymos, 2025-02-19:
https://github.com/L10N37/PSXTAL if it's a PU-7 I'm not sure it works.

godmanpicoli, 2025-02-20:
> https://github.com/L10N37/PSXTAL se for um PU-7, não tenho certeza se funciona.

For American games, is it necessary to change the internal clock?

kalymos, 2025-02-20:
Sorry I have never used this mod, I am primarily concerned with psnee. but I think I have already heard of this problem, only I do not remember in which forum, maybe psxdev

godmanpicoli, 2025-02-22:
> Sorry I have never used this mod, I am primarily concerned with psnee. but I think I have already heard of this problem, only I do not remember in which forum, maybe psxdev

I replaced the Arduino Nano 328PB with an Arduino Pro Mini 328P and it worked normally.

kalymos, 2025-02-22:
Glad it works for you, but I thought the only difference between these two was the package. 
I absolutely have to do a global test campaign again.
by the way thanks for the screenshots it helped me understand what is used to detect the BIOS.

closed 2025-03-19 by kalymos

ba114, 2025-03-27:
> > Sorry I have never used this mod, I am primarily concerned with psnee. but I think I have already heard of this problem, only I do not remember in which forum, maybe psxdev
> 
> I replaced the Arduino Nano 328PB with an Arduino Pro Mini 328P and it worked normally.

Are you saying that switching to the pro mini 328P with no other changes now allows you to play PAL and NTSC-U games on your SCPH-1000?

I have the 40-pin bios SCPH-1000 with pro mini 328PB and it only plays NTSC-J discs + backups. No other regions.

Is yours pro mini a 5v 16MHz model or 3.3v 8MHz model?

kalymos, 2025-03-28:
I just received 328PB cards yesterday, and they work fine, moreover if you look on the first page you will see that I have updated the table of tests that I have done.
For ATmega cards, a 16MHz crystal is essential.

ba114, 2025-03-28:
> I just received 328PB cards yesterday, and they work fine, moreover if you look on the first page you will see that I have updated the table of tests that I have done. For ATmega cards, a 16MHz crystal is essential.

For scph-1000 40pin with PAL and NTSC-U?

kalymos, 2025-03-28:
Yes, I tested this combination yesterday morning. In fact, with that, I was able to update the table of tested combinations. 
https://github.com/kalymos/PsNee/blob/master/images/test-PSNee-v8.6.png

If you want me to help you more, I need to.
The motherboard number. 
The code injection technique you used.
 A photo of the solder joints on the BIOS.

ba114, 2025-03-28:
Think I killed my ps1 anyway by being lazy and trying to flash my pro mini
without disconnecting from the ps1

kalymos, 2025-03-28:
Unfortunately, this is one of the biggest risks, especially on older models. If you want to start again with another one, choose the later models, SCPH-700x-7500x-900x-10x, they are much more robust, and the drive is of better quality.
At first I made the same mistake, luckily it was on a SCPH-102 model, which held up.