kalymos/PsNee #71: PSNeecore-1.2 folder goes where?

nex86, 2025-03-15:
Where exactly do I put this folder?
documentation mentions "arduino/hardware/" but where exactly?

I found C:\Users\nexxu\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
has these files, but do I replace them?

I'm an absolute noob with Arduino IDE.

kalymos, 2025-03-15:
When you install the Arduino IDE, you'll normally see the Arduino folder in your home directory, where you'll find a Sketch folder where your programs are saved by default. This is the folder.

nex86, 2025-03-15:
The folder in documents?
There is a library folder and a MAX30100_adafruitio  folder

nex86, 2025-03-15:
Okay.. I now put it in 
Documents\Arduino\hardware\PSNeecore-1.2\avr

kalymos, 2025-03-15:
Yes something like that, and normally the next time you open Arduino IDE it should appear in Tools -> Board

kalymos, 2025-03-15:
P.S. The other path is possible, but it's more complicated to explain.
P.P.S. Once you've successfully installed your application, you can answer the short questionnaire in the discussion section.
Have fun!

nex86, 2025-03-15:
Ok got this but now I have more issues.

its getting me "request upload failed" 
when compiling and uploading the sketch.
I have an Arduino Micro 32U4

kalymos, 2025-03-15:
Are you at step 4 ?

nex86, 2025-03-15:
yes but tools -> varient is also something that is not listed on there.

kalymos, 2025-03-15:
Tools -> Processor -> your MCU varient for the modchip
It's corrected, excuse me.

nex86, 2025-03-15:
Ok so I picked:

Board: PSNEE->ATMEGA32u4
Port: COM6 (Arduino Micro / ATMEGA32u4
Programmer: Arduino as ISP (ATMEGA32u4)

when I know upload the sketch it keeps on "uploading.."
and the LED on the Arduino is blinking. 
How long is it supposed to take?

kalymos, 2025-03-15:
Not long, normally the program tells you when it's finished injecting.
Oh yeah, that's right, your target doesn't have the blink LED itself.
Don't worry, I'm interested in hearing about your problems; it lets me know what mistakes I've made.

kalymos, 2025-03-15:
I edited this, I hope it's understandable.

For ATtiny, and ATmega32U4 diagram, and modify the Blink sketch accordingly

    ATtiny LED_BUILTIN -> PIN 3
    ATmega32U4 LED_BUILTIN -> PIN 10

nex86, 2025-03-15:
I tried it with original Arduino Micro and Pro Micro that has the USB-C , but both behave the same.

I'm no sure if I'm doing this wrong here.
Do I need two Arduinos to program one?
from my Understandig, it is only required when the microcontroller does not have a programmer by its own.
But my Micros do have a programmer.

kalymos, 2025-03-15:
I see, yes to program in ISP you need two cards, the first which becomes the ISP programmer, and the second which will be your target.

kalymos, 2025-03-15:
If you don't need the bios patch, you can just use USB injection

nex86, 2025-03-15:
According to the guide, SCPH 9002, does not need bios patching. but correct me if I'm wrong.
USB injection does not seem to work. or do I need to change anything?

kalymos, 2025-03-15:
Give me half an hour to check if the USB injection works properly. I stopped using it for my tests.
PS: I still have to tweak the ISP injection tutorial a little. In the meantime, if you find anything you don't understand, let me know.

nex86, 2025-03-15:
I tried to inject with my Arduino Uno, connected it as said in the diagram but when I go on "upload using a programmer" it says "failed."

kalymos, 2025-03-15:
USB injection works very well for me.
What compilation options do you have comments in the sketch?

Normally when you have installed a chip on your console and it works at least partially with the LED_RUN option, the letter that flashes with the blink sketch will light up with each injection of the region code.

nex86, 2025-03-15:
It looks like I can not upload anything as usual anymore. 
The Arduino Micro disconnects itself just before uploading.
It did the ISP Gateway method with my Uno, and USB uploading broke after this.

kalymos, 2025-03-15:
It's not exactly a great idea.
It could be that the boards you have are no longer working properly, have lost their bootloader, or that the fuses are no longer working properly.
There's a fairly simple solution for this: reinstalling a bootloader using an ISP gateway.
In the tutorial, there's a link that explains more about the intricacies of ISP gateways. https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP/

nex86, 2025-03-15:
I ended up using the Micro pro, and it works.
In case I have to update the firmware, what steps do I need to do?
Can I do this right while its connected to the motherboard of the PSX or do I have to desolder it?

closed 2025-03-19 by kalymos