kalymos/PsNee #72: Is it worth updating my existing PsNee modchip?

dazzaXx, 2025-03-18:
Hello, I installed a PsNee modchip in my scph-1000 Japanese console sometime near the start of last year and noticed a lot to do with the code has changed since including the bios patching.

Would it be worth it to update my existing PsNee to the latest version with bios patching or just leave it as is?

Thank you!

kalymos, 2025-03-18:
Hello
Basically no, the operation is identical.
The main change is that I put back MCUs that we had lost along the way, and
that there is a unified way of injecting the code via ISP

dazzaXx, 2025-03-18:
> Hello
> Basically no, the operation is identical.
> The main change is that I put back MCUs that we had lost along the way, and
> that there is a unified way of injecting the code via ISP
> 
> Le mar. 18 mars 2025 à 23:54, dazzaXx ***@***.***> a écrit :
> […](#)

No worries, thank you!

closed 2025-03-18 by dazzaXx