kalymos/PsNee #74: NTSC-J 7000 memory card

Nightalex77, 2025-03-24:
The new version does not solve the problem with the memory card manager on the 7000 model NTSC-J? When trying to log in, the console freezes and reboots after 30-40 seconds.

kalymos, 2025-03-24:
Good morning,
If you are using v8, in principle the behavior is the same.
We will have to do some tests to confirm this.

Nightalex77, 2025-04-09:
Try entering the memory card menu without a game disc.

kalymos, 2025-04-12:
I took the time to do some tests.

What I've personally noticed with the current version is that PlayStations with BIOS version 4.0j are indeed unable to load memory cards. And it's possible that this could cause your system to crash.

All other versions I've tested don't have this problem.

The solution I see at the moment is to uncomment the #definePATCH_SWICHE option and install a switch like the diagram you'll find in the wiki.
With this solution, you'll be able to handle memory cards by disabling the BIOS patch.

Sylph4, 2025-05-20:
Have the same issue on 7000 J model. Otherwise works ok.

kalymos, 2025-05-29:
I've debugged the #define PATCH_SWITCHE build option and updated the wiki diagram.

To resolve this memory card access issue, or simply disable the BIOS patch, simply undo #define PATCH_SWITCHE in the options section of the PSNee***.ino file and install a switch as shown in the PIN_SWITCH diagram available on the wiki.

thiagosepp-x, 2025-09-06:
So sad, today i have the same problem. No memory card manager and Cd Player resets the console. Maybe the future reveals a solution better than a Switch.