kalymos/PsNee #78: How do you upload via usb to a pro micro?

dazzaXx, 2025-03-31:
What board type is used for a pro micro? I tried the atmega32u4 option included with the core files but I kept getting a upload failed error. Thanks!

kalymos, 2025-03-31:
Hello,
I'd need a little context to answer your question.
Did you want to inject the code via USB or via ISP?
Is it possible to have a copy of the error message?
Perhaps also the serial number of your console and your motherboard?

kalymos, 2025-03-31:
Have you tried using PSNeecore?
Or is it the standard card manager?

What option did you choose (Tools -> Programmer ->)?

dazzaXx, 2025-03-31:
Hi, I'll close this as it seems it isn't a PsNee issue, it must be the PU-8 motherboard itself, I've tried even with a properly flashed Nano via ISP with no luck at all, compared to my PU-22 that works flawlessly with the nano.

kalymos, 2025-03-31:
Often on old consoles the CD player is a bit out of breath, for that on the PSNee side you can perhaps slightly increase the HYSTERESIS_MAX variable for example around 18-20, and slightly decrease the DELAY_BETWEEN_INJECTION delay. On the hardware side cleaning and lubricating with silicone grease the plastic friction parts of the player can be useful, if the motor makes a nasty noise during rotation speed changes a drop of oil in it or WD-40 can raise it from the dead.

dazzaXx, 2025-04-07:
> Often on old consoles the CD player is a bit out of breath, for that on the PSNee side you can perhaps slightly increase the HYSTERESIS_MAX variable for example around 18-20, and slightly decrease [...]

Thank you for the tips! I've got some spare nanos, their usb chips must be bad but they still function as they should with the blink test flashed via an ISP. I'll give what you suggested a go and see if I can get it working. Thank you!

kalymos, 2025-04-07:
Regarding the HYSTERESIS_MAX variable, I updated the project by increasing it to 17. I've been running some tests on it over the past few days, and I've also updated the comments for this variable.
As for the DELAY_BETWEEN_INJECTION variable, I don't think there's any point in modifying it.
And finally, for cleaning, the WD isn't the best option; I'd need to write a complete tutorial for that.

dazzaXx, 2025-04-07:
> Regarding the HYSTERESIS_MAX variable, I updated the project by increasing it to 17. I've been running some tests on it over the past few days, and I've also updated the comments for this variable. As for the DELAY_BETWEEN_INJECTION variable, I don't think there's any point in modifying it. And finally, for cleaning, the WD isn't the best option; I'd need to write a complete tutorial for that. P.S. about the blink test in ISP, I do it because arvdue even if it tells you that it injected correctly, in reality it does not know it, it only knows that it was able to transmit the program to the ISP gateway, the only way to see if it goes well is to see a program running on the target.

Ok thank you, I'll download the latest version of PsNee, I did the blink test on both nanos with bad usb chips and they both ran the blink test fine, I'm planning on removing the usb port anyway to make the pcb not as tall, since it also doesn't work as well.

kalymos, 2025-04-07:
By the way, a little thing about the nanos that have problems with USB, it could be that they no longer have a bootloader, but we can reinject it with the ISP in the Arduino IDE.
Tool >> Burn bootloader

dazzaXx, 2025-04-07:
Update, I've tested several arduinos with this PU-8 board, but it seems at least some connection is wrong or non functioning, I have noticed that the LED that lights up when injecting doesn't light up at all. Only when the board has power.