kalymos/PsNee #79: PU-8 E35D SUBQ Pin diagram is incorrect?

dazzaXx, 2025-04-07:
Hi, me again. I figured out the issue with my PU-8 board with the big chip. It's setup like the PU-7, with it being the pin above the one shown in the PU-7 diagram.

[image]

kalymos, 2025-04-07:
Thanks for the info.
Is it more like this?
[image]

kalymos, 2025-04-07:
I think I don't have this variant, can you tell me a good point for the reset?

dazzaXx, 2025-04-07:
> I think I don't have this variant, can you tell me a good point for the reset?

How would I go about finding this? My PU-8 Board is from a SCPH-1002.

Also that diagram is correct, yes.

dazzaXx, 2025-04-07:
To confirm my wiring:

[image]

kalymos, 2025-04-07:
I think it's the same as the other variant of the PU la-8.
To check, you need a multimeter. Put it in continuity mode, one probe on the GNT and the other on the RESET. If continuity is present when you press the hood closing switch, that's the correct point.

dazzaXx, 2025-04-07:
> I think it's the same as the other variant of the PU la-8. To check, you need a multimeter. Put it in continuity mode, one probe on the GNT and the other on the RESET. If continuity is present when you press the hood closing switch, that's the correct point.

I'll have a look and report back

dazzaXx, 2025-04-07:
Hmm, I can't seem to find it. I've tested a few points in and around the area but I've got nothing from them

kalymos, 2025-04-07:
Sorry for the reset, it's connected to the top pin of the power connector.
[image]

kalymos, 2025-04-07:
Thanks again, I will update the diagram.

kalymos, 2025-04-07:
[image]

dazzaXx, 2025-04-07:
> [image]

Awesome, that's it. Glad to have helped 🙂 I'll close this issue now, thanks again!

closed 2025-04-07 by dazzaXx