kalymos/PsNee #80: Hit or miss booting NTSC backups on SCPH-102

RutgerKroeze, 2025-04-17:
I have installed version 8.6 on a Pro Mini and soldered the chip in my SCPH-102. The console boots PAL backups fine, 100% success rate. But NTSC backups are very dodgy. The console is at the main BIOS screen and the CD-reader spins up, then spins down, spins up, spins down etc. I verified the backups work fine in another chipped PS1.

At first I thought it was because of the wire length in my install, I placed the Pro Mini in the upper right corner of the board.
[image]

So I've relocated the chip to the left of the Alliance branded DRAM chip on the board and switched from solid core 28AWG Kynar wire to stranded 28AWG silicone wire. Power wires are thicker, not sure this is necessary.
[image]

But the behavior is exactly the same. Any ideas what can cause this? Could the BIOS patch be the culprit?

kalymos, 2025-04-17:
I've never encountered this particular problem, but we'll see what we can do.

One thing is certain, it's not the length of the cables. On my test models, I have harnesses with 20 cm cables to have a connection harness outside the console, and there is no problem.

I thought for a moment that it might be the player, but since the European copy disc plays fine, it must not be that.(Note that it's never a bad thing to give them a service. On plastics, use only silicone grease.)

Did you inject the code via ISP?
If you injected into iSP, what method did you use?
What is the reference of the ATmega 128/328 P/PA/PB?

RutgerKroeze, 2025-04-17:
I've already serviced the CD unit. Cleaned the lens with IPA, cleaned and regreased the rails. It's less noisy now and there's no FMV stutter, even with (PAL) backups. When NTSC backups boot, they run fine as well. So it definitely seems to be related to the modchip.

I used an Arduino Uno with the Arduino as ISP sketch to inject the code. That seemed to go well.

The chip on the Pro Mini (from AliExpress, so a clone?) reads MEGA328P.

Hope we can find something out! If I remember correctly, I started out with postal2201's PSNee V8 repo on the first attempt. Your repo was still at V7 back then. I've reinstalled the code from this repo last week, no difference.

kalymos, 2025-04-17:
There is one last thing that may possibly have an influence, it is the fuses of the ATmega, how did you put them? (personally I use this variation H: DF, L: EE, E: FD)

RutgerKroeze, 2025-04-17:
I didn't do anything to the fuses when I updated to V8.6 from this repo. Previously I had set it to H: DF, L: FF, E: FF, as instructed by postal2201's repo. I think..

What's the best method to verify this? Would it help to set the fuses to your suggestion?

kalymos, 2025-04-17:
Maybe. Look in the wiki, I explain my method with the project's board manager, which sets the fuses automatically.

https://github.com/kalymos/PsNee/wiki/Personal-recipe-to-make-a-good-PSNee

RutgerKroeze, 2025-04-17:
The fuses must be correct then because I followed those steps from the wiki! So installing the PSNeeCore file and selecting the target MCU via PSNeeCore takes care of the fuses? 

I will double check those settings and reprogram the Pro Mini, just in case. Guess I just have to disconnect the VCC wire between the modchip and PS1 to avoid injecting 5V into the PS1 board? The Pro Mini is a 16MHz 5V variant, does that matter anything?

RutgerKroeze, 2025-04-20:
I've uploaded the latest version with HYSTERIS_MAX parameter at 17, makes no difference. I think it's related to the BIOS patch for SCPH-102. The purpose of this patch is to bypass the additional region locks in the BIOS, right? Given that PAL backups boot consistently, while NTSC disks almost never boot..

The BIOS patch has changed since V8, correct? Up until V7 the PAL PSone BIOS patch seems to be the exact same patch as implemented in the old ONEchip, at least it uses the same pins. Shall I try the V7 branch to see if it makes a difference? Of course there could be something wrong with my Arduino as well..

kalymos, 2025-04-20:
I haven't done any advanced testing with this patch version to see if it's necessary to pass both PAL games, specifically on this console model. Since I have the same one and it works fine for me.
But from memory, even PAL games won't pass without the bios patch.
Always try version V7, because as you assume, the biopatch doesn't work the same way on version V8.
Let me know if it works for you, if it works I'll see if I can make a legacy bios patch.
V7 is categorized into project branches.

RutgerKroeze, 2025-04-21:
I've uploaded the V7 branch and changed the two BIOS pins. Working very well now! I can see that the BIOS stays in NTSC443 mode on my upscaler. I've tried multiple backups, both PAL and NTSC and all of them boot fine, 100% success rate. Only a few times after power off and power on again, the BIOS switches into PAL mode when the white Sony logo appears, indicating the BIOS patch failed. My guess is the BIOS patch works 90% of the time, but it could also be that I'm powering on too quickly after power off.

Maybe there's something wrong with my Pro Mini, maybe it's my particular console, maybe it's both. But fact is that the 'old' NTSC BIOS patch from V7 is working much better for me. I'll stay on this branch because I don't need any of the added features of V8+ and I'm not using the portable PAL screen.

Thanks for your help and feedback!

kalymos, 2025-04-21:
Thanks for the feedback.
I'll see what I can do to integrate this version of the patch.

closed 2025-04-21 by kalymos