kalymos/PsNee #81: Game Discs Stop Working After Console Reset

Raz0r029, 2025-04-21:
I recently flashed version 8.6.2 onto an Arduino Nano and installed it in my PU-7 1.1J BIOS SCPH-3500 console. While all backup and original game discs from any region work without issues, when I reset the console, it fails to read the disc and just returns to the main menu. I still have to power cycle the console just to read game discs again. This problem happens only with NTSC-U and PAL discs, while NTSC-J discs load fine even resetting the console. Is this normal, or could it be a bug?

kalymos, 2025-04-21:
This strongly resembles a small problem with the reset line, which is desynchronizing the BIOS patch.
First possibility: there's a problem with your RESET cable; check the solder joints and the solder point.
Second possibility: the voltage levels that trigger the reset are too different between the PlayStation and the Arduino board. In these cases, pressing and (holding) the reset button can have a positive effect.

Raz0r029, 2025-04-21:
I replaced the reset cable and resoldered it properly, but the problem still remains. Holding the reset button for 5–10 seconds didn’t help either. I even tried to solder the reset cable directly to pin 7 on the PSU, but it made no difference.

kalymos, 2025-04-21:
If you can read copied discs, it is because you have normally compiled with this option #define SCPH_3000 ?
I'm going to do some tests on a console with the same bios.

Raz0r029, 2025-04-21:
Yes, I used #define SCPH_3000 instead of #define SCPH_3500_5000 because the latter is meant for the 2.1J BIOS, while mine is a 1.1J.

kalymos, 2025-04-21:
I must admit I don't know where this is coming from. I tested it on my hardware and I can't reproduce this bug.
When you have problems with the timing of the biopatch injection, most of the time the console won't start.
I'm currently testing version 8.7; it has some optimizations for region code injection. You can try it out if you want (I tested it on a PU-7 1.1j console), but I can't guarantee that this will fix your console's problem.

Raz0r029, 2025-04-21:
I see. I suppose I can try version 8.7 and see if that fixes the problem. Also, if I use an ISP programmer to flash the chip while it's still on the motherboard, do I need to desolder all the wires first, or can I connect the programmer without removing them?

kalymos, 2025-04-21:
Theoretically, it's better to bury the unsoldered terminals, but at the very least, solder the VCC. For ISP injection, if you use my card manager, there's no problem with the fuses. If you use another solution, remember to configure the fuses.

kalymos, 2025-04-22:
Thanks for reporting that you finally managed to do something.
For v8.7, it's mostly minor optimizations, and an older bios patch that works better for certain conditions for the 102s. I'm at the stage "coups de savate pour faire rentrer toute le tout"

closed 2025-04-23 by Raz0r029