kalymos/PsNee #83: PSOne will not read burned discs

impy101, 2025-05-08:
I have tried several modchips, such as Mini pro and Nano, but none of them allow me to play burned discs on my PSOne (PM-41 PAL). I followed the guide to the letter, programmed all chips using uno as programmer and the blink program worked fine. So at least that part should be solid.

I checked the wiring 4 times and that, too, seems to be spot-on. I just get a screen that shows minutes, but it will not read. 

Any help would be much appreciated.

kalymos, 2025-05-09:
We'll see what we can do.
If I understand correctly, your console is a SCPH_102.
But to move forward, I need some additional information: whether you followed the guide for injection via USB or via iSP.
And which option did you uncomment in the .ino file?
Does the Arduino LED flash like in the gif in the tutorial you followed?

impy101, 2025-05-09:
> We'll see what we can do. If I understand correctly, your console is a SCPH_102. But to move forward, I need some additional information: whether you followed the guide for injection via USB or via [...]

I injected the code via ISP and I did flash the blink code to see if it worked and yes it did. 

I uncommented: #define ATmega328_168  and #define SCPH_102

kalymos, 2025-05-09:
What I wanted to know about the blinking LED, once the mode is installed in the console and you test a disc does the LED react like the gif at the end of the tutorial?

P.S. The boards you tested on currently no longer have a bootloader, you can reinstall them via ISP in the Arduino.

impy101, 2025-05-09:
> What I wanted to know about the blinking LED, once the mode is installed in the console and you test a disc does the LED react like the gif at the end of the tutorial?
> 
> P.S. The boards you tested on currently no longer have a bootloader, you can reinstall them via ISP in the Arduino.

I currently have a Mini pro installed and there are 2 solid red lights. No blinking when a burned disc is spinning up.

kalymos, 2025-05-09:
This suggests that the BIOS patch isn't receiving the correct triggers and is stuck.
It's possible to have a photo of your setup so you can see the connections and soldering.
Basically, PSNee has two phases of operation.
The first phase observes a memory address pin AX to determine when to modify a BIOS bit by operating on one of the BIOS memory data output pins DX. In this state, the LED remains lit until the triggers are reached.
The second phase reads the data stream between the CD controller and the processor to inject the region code. In this phase, the LED only lights up during the injection.
In other words, there may be a bad connection between the BIOS address input and pin 2 of the Arduino.

impy101, 2025-05-09:
[image]

Here you go

kalymos, 2025-05-09:
I think I found a problem, the wire you connected to pin 3 of your Arduino, should be connected to pin 2.

[image]

kalymos, 2025-05-09:
We've made a little progress, which is already good.
So now, when you play a burned CD, which screen do you arrive at? The PlayStation menu, the warning that it's not a genuine disc, or the region confirmation screen (PlayStation logo on a black background)?
Does it flash constantly now, or only three or four times?

kalymos, 2025-05-09:
And I just thought of two other things, does it have a 16mHz crystal, it seems to me that this pin is badly soldered.

[image]

impy101, 2025-05-09:
> We've made a little progress, which is already good. So now, when you play a burned CD, which screen do you arrive at? The PlayStation menu, the warning that it's not a genuine disc, or the region confirmation screen (PlayStation logo on a black background)? Does it flash constantly now, or only three or four times?
> 
> P.S. The soldering of the data seems a little weak to me. and I can't see the SUBQ SQCK soldering

I've touched up the data point. Solid connection to the capacitor now. However, no change. In fact, after changing the wire to pin 2 it's become quite buggy. Sometimes I just get a black screen, other times I see the playstation screen in grey and when it does start up properly I just see either that screen with minutes on it or a red screen saying I need to put in a playstation disc (depends on which cd I put in)

impy101, 2025-05-09:
I've also counted the number if flashes and I've counted 7 flashes.

impy101, 2025-05-09:
Ive now got new problem where it starts up properly but boot loops on its own. 

You know what, I'm going to remove the mini pro and put in a brand new nano and see what it does then.

kalymos, 2025-05-09:
If you plan on removing the board, check all the soldering on the PS carefully. It's not the easiest board to model.

impy101, 2025-05-09:
Put in a brand new Nano and now it works flawlessly... I guess not every arduino is created equal.

kalymos, 2025-05-09:
I've tried a lot of Chinese variations, and with this one, if there was a problem, it was me ;)

closed 2025-05-09 by kalymos