kalymos/PsNee #86: Scph-9001 psnee install now causes the spindle motor to move rapily even when lid sensor is not pressed down

Genomans, 2025-05-23:
- <html><body>

Console Reference | SCPH_9001
-- | --
Motherboard reference | PU-22
MCU referance | 328P - 16MHz
Injection method | PSNeeCore - Upload through USB
The option you uncommented in the code | #define SCPH_xxx1 #define SCPH_7000_9000 
BIOS version | not sure
Basically how the LED flashes | solid red light on pow
A description of the problem encountered | I installed it wrong i thought it was like the 7v but with bios solder with it now the console won't read games and the spindle motor rapilly spins on startup even when lid is open
And a photo of your installation | [image]

</body>
</html>

kalymos, 2025-05-23:
First, you just need to uncomment #define SCPH_xxx1
And you can unsolder the D2, D3, and D4 cables, since American consoles don't need the BIOS patch.

kalymos, 2025-05-23:
Do you service the optical unit motors (WD40 does the job for this), and clean the old lubricant on the friction areas (replace with silicone grease)?

Genomans, 2025-05-23:
yes i did  service the motor and laser a relubed the plastic and cleaned the laser using iso alcohol i the laser reads some games back then now it does not after the install. I adjusted the pots and nothing still happends

kalymos, 2025-05-23:
Normally, there's no point in touching the potentiometer on the optical block (it's actually pretty crap).
Does the problem persist even when nothing is installed?

Genomans, 2025-05-23:
yes it does like a said in the start I desoldered it and the spindle motor still does that but after maybe a day or two it goes back too normal 2 spins lens lifting. but after leaving it on for maybe about 30 minutes the spindle acts up again and a have to wait another day or two just for it to go back to normal for 30 mins.

Genomans, 2025-05-23:
also another note this was been a thing since feb ever since your ver of psnee was public i just the time to report it now since a was busy with other projects. thats why i know that the spindle acts normal for a short while

Genomans, 2025-05-23:
no problem it just that this behavior is very unique and im just confused why this happened maybe its my botch soldering

Genomans, 2025-05-24:
will do

david4599, 2025-06-07:
> I installed it wrong i thought it was like the 7v but with bios solder with it now the console won't read games and the spindle motor rapilly spins on startup even when lid is open

The fact that the motor spins when the lid is open tells that psnee is not the issue here but the wrong installation probably caused a sort of short circuit or something somewhere...

Also (which is why I initially wanted to reply):

> Do you service the optical unit motors (WD40 does the job for this)

By experience, WD40 is not a good product to properly lubricate the spindle and sled motors (only a short term solution and this will be actually worse with too much of it, ask me how I know...).
Light oil should be used instead. I highly recommend checking this youtube video to know how to apply the oil (even though I prefer applying it without removing the original ceramic spindle hub which is fragile and can break easily like he says):
https://www.youtube.com/watch?v=cnNTnlaD3rI

Genomans, 2025-06-08:
Could be gonna check that out once I have time